OpenNuvoton/M480BSP #1: MPU gives Memory Fault !! all the time when running MPU sample code

rdhobi, 2018-08-16:
Hello,
We are using Nuvoton PFM-M487 kit and running `SampleCode/CortexM4/MPU/main.c`. MPU always gives Memory Fault!! without pressing any key. Is there any configuration missing in sample code?

yachen, 2018-08-17:
Hi @rdhobi ,

Please pull commit 1d9cba1 to fix this issue. Thanks.

Sincerely,

Yi-An Chen

rdhobi, 2018-08-17:
@yachen: Thanks, we have tried it yesterday and example runs fine. But when MPU give "Memory Fault !!"? Accessing memory in region should give Fault.

yachen, 2018-08-17:
Hi @rdhobi ,

I don't understand your question. Are you asking why previous code triggers memory fault?

Sincerely,

Yi-An Chen

rdhobi, 2018-08-17:
@yachen No, I mean to say that if I access memory inside locked region, then MPU should give memory fault right?

For example, we have configured region 3 with base address 0x20004000 with 1KB size and if I will access
0x20004000 memory then MPU should give memory fault. 
If I'm wrong then when should MPU give memory fault?

yachen, 2018-08-17:
Hi @rdhobi ,

That's right, you should see memory fault while access 0x20004000.
The console output looks like below now:

 Start MPU test:

 Please Press '1' to read memory from region 1 (Flash Memory)
 Please Press '2' to read memory from region 2 (SRAM)
 Please Press '3' to read memory from region 3 (Test Memory)
 Memory Fault !!

Sincerely,

Yi-An Chen

rdhobi, 2018-08-17:
@yachen Thanks for your response but I don't see any Memory Fault!!. MPU is allowing all the memory access from region 1,2 and 3. Can you please check?

yachen, 2018-08-17:
Hi @rdhobi ,

What I paste on previous response is the actual output from UART console.
This sample can trigger memory fault as expected.

Sincerely,

Yi-An Chen

rdhobi, 2018-08-20:
@yachen: You should also get memory fault for region 1 and 2 memory access right?

yachen, 2018-08-20:
No, region 3 only.

rdhobi, 2018-08-20:
@yachen Got it. region 1 and 2 not giving Memory Fault!! due to AP_Pri_RW_User_RW permission. right?

rdhobi, 2018-08-23:
@yachen I have configure region four with AHBPERIPH_BASE(0x40000000) as starting address and region size is 64K. Now I'm accessing LED which is GPIO having base address is 400041C0. During toggling GPIO MPU give memory fault but still I can turn on that LED using same GPIO. Can you please help? MPU configuration is below
```
MPU->RBAR=((0x40000000 & MPU_RBAR_ADDR_Msk) | (0x4 & MPU_RBAR_REGION_Msk) | MPU_RBAR_VALID_Msk);

MPU->RASR = (( AP_No_Access<< MPU_RASR_AP_Pos)| ( Region_Size_64K << MPU_RASR_SIZE_Pos) | MPU_RASR_ENABLE_Msk);

SCB->SHCSR = SCB_SHCSR_MEMFAULTENA_Msk;

MPU->CTRL |= MPU_CTRL_PRIVDEFENA_Msk | MPU_CTRL_ENABLE_Msk;
```

```
 void LED_Init (void)
{
	M32(&PH->MODE) = 0x00005555;           /* LEDs on PORT E defined as Output   */
	PH->DOUT |= 0x000000FF;                /* switch off LEDs                    */
}

void LED_On (unsigned int num)
{
    if (num <8)
    {
        PH->DOUT  &= ~led_mask[num];
    }
}

```

rdhobi, 2018-08-31:
@yachen: Can you please help?

yachen, 2018-08-31:
Hi, we're checking.

rdhobi, 2018-09-08:
@yachen: We have tested same code on other MCUs having MPU. LED will not on after locking it from MPU region.

yachen, 2018-09-20:
Hi, 

The memory fault interrupt is triggered before the illegal memory access take place.
For M480, you'll have to reset the system in memory fault handler to prevent the illegal access.

Sincerely,

Yi-An Chen

yachen, 2018-09-27:
@rdhobi Oops. In order to let sample code keep executing, we clear the MPU->CTRL register in MemManage_Handler(). Don't clear this register if you need to block the illegal memory access. Sorry for the confusion.

rdhobi, 2018-09-27:
@yachen: Thanks for reply, this is our main confusion like why MPU is disabled MemManage_Handler() in sample code. Now its clear, but you should make add comments  for avoiding confusion.

yachen, 2018-09-28:
Yes, we'll update the comment. Thanks.

closed 2018-09-28 by yachen