OpenNuvoton/M480BSP #2: FMC OTP Write and Read

rdhobi, 2019-01-25:
Hi,
I am using nuvoton m487JIDAE controller. I took reference from https://github.com/OpenNuvoton/M480BSP/tree/master/SampleCode/StdDriver/FMC_OTP code.
I wrote data in specific FMC_OTP address using FMC_Write_OTP API and Read data from same FMC_OTP address using FMC_Read_OTP API, it gives me proper value. But when I over-write data in FMC_OTP address, it gives me wrong value.
Is it possible to overwrite the data in the memory which is not locked?
I haven't enable FMC_OTP_Locked bit for the same address yet.
[FMC_OTP_LOGS.txt](https://github.com/OpenNuvoton/M480BSP/files/2795409/FMC_OTP_LOGS.txt)

yachen, 2019-01-25:
Hi Rahul,

If OTP is not locked, you can overwrite OTP by changing 1 to 0, but not vice versa.
Since this is OTP, we suggest you just write it once and lock it.

Sincerely,

Yi-An Chen

===========================================================================================
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

yachen, 2019-04-18:
Close since there's no further discussion for a while.

closed 2019-04-18 by yachen