OpenNuvoton/M480BSP #4: I2C_Wakeup_Slave is not working

chengsg78, 2019-06-26:
The  I2C Wake-up interrupt flag wont be set, when receive i2c address match.
 I2C_STATTUS1->ADMAT0= check.

chengsg78, 2019-06-27:
I have check the code without modify, the code is not working,
 I2C_GET_WAKEUP_FLAG(I2C0) is unbale to get true.

    if(I2C_GET_WAKEUP_FLAG(I2C0))

    {

        /* Clear I2C Wake-up interrupt flag */

        I2C_CLEAR_WAKEUP_FLAG(I2C0);

        g_u8SlvI2CWK = 1;

        return;

    }

chengsg78, 2019-06-28:
/* Check I2C Wake-up interrupt flag set or not */
    if(I2C_GET_WAKEUP_FLAG(I2C0))    // this wont not able to set

mjchen1, 2019-07-02:
I test the sample code.
It looks like work normal.
[image]

closed 2019-08-16 by chengsg78