OpenNuvoton/M480BSP #6: Will ISP examples for GCC be available soon?

adrianyuenmi, 2020-03-05:
It would be ideal to also develop ISP firmware for LDROM without needing Keil

danchouzhou, 2021-08-23:
Hello adrianyuenmi, 

I'm working on a open source USB audio codec project with Nuvoton's chip. I need a ISP so that user does not need a programmer to update firmware in the APROM, so I do a porting of ISP_HID for GCC.

All the things I do is: 
1. Modify __align(4) into __attribute__((aligned(4)))
2. Remove unuse interrupt vector in the startup file.
3. Additionally I add some code let it use HIRC48M if the chip is support.

By ussing gcc-arm-none-eabi-10.3-2021.07 we get 3868 bytes of code size. Also please refer to the following link.

[](https://github.com/danchouzhou/NuDAC/tree/0c574726f2368c9df219f9dd103b413c1e14eba2/firmware/ISP_HID)

Cheers, 
Danny

ZaleYu, 2021-08-23:
Hi Danny:
Could you pull a request on M480BSP?
Your effort will make more customers appreciate. 
Thank you!

Sincerely, 
Zale Yu

danchouzhou, 2021-08-23:
Hello Zale Yu, 

Please refer to [Pull request #22 ](https://github.com/OpenNuvoton/M480BSP/pull/22). 
The code can be compiled, but has not been tested on target device. Please arrange someone to test, thank you!

Cheers, 
Danny

ZaleYu, 2021-08-24:
Hi Danny:
Yi-An Chen is the owner of M480BSP. He will make a response.

Thank you.

Sincerely, 
Zale Yu

yachen, 2021-08-25:
Engineers maintain ISP code are WFH without board in hand. But I think we can start testing from next week.

closed 2021-12-09 by ychuang3