OpenNuvoton/M480BSP #7: 例程："HSUSBD_Mass_Storage_SD" 不能在192MHz主频下工作。

stankong, 2020-04-01:


stankong, 2020-04-01:
单独使用SD卡例程和HSUSBD_Mass_Storage_SRAM例程都可以在192MHz下正常工作，HSUSBD_Mass_Storage_SD例程也能在50MHz下正常工作，但是后者核心频率提高成192MHz就无法发现USB设备。。。

stankong, 2020-04-08:
自己解决问题了，注意到异常时HUSBD_BUSINTEN寄存器会和写入结果不一样，怀疑192MHz频率下HUSBD模块不能仅靠验证HSUSBD->EP[EPA].EPMPS寄存器来确认稳定。修改HSUSBD_Open函数，增加判断此寄存器写入检验代码，实验表明可以正确工作在192MHz下了。

```
    while (1)
    {
				HSUSBD_ENABLE_BUS_INT(HSUSBD_BUSINTEN_RESUMEIEN_Msk|HSUSBD_BUSINTEN_SUSPENDIEN_Msk|HSUSBD_BUSINTEN_RSTIEN_Msk|HSUSBD_BUSINTEN_VBUSDETIEN_Msk);			
				if(HSUSBD->BUSINTEN == (HSUSBD_BUSINTEN_RESUMEIEN_Msk|HSUSBD_BUSINTEN_SUSPENDIEN_Msk|HSUSBD_BUSINTEN_RSTIEN_Msk|HSUSBD_BUSINTEN_VBUSDETIEN_Msk))
					break;
    }
```

closed 2020-04-08 by stankong

stankong, 2020-04-09:
进一步优化的补丁，
```
diff --git a/Library/StdDriver/src/hsusbd.c b/Library/StdDriver/src/hsusbd.c
index eb13e9a..4e1f885 100644
--- a/Library/StdDriver/src/hsusbd.c
+++ b/Library/StdDriver/src/hsusbd.c
@@ -76,6 +76,10 @@ void HSUSBD_Open(S_HSUSBD_INFO_T *param, HSUSBD_CLASS_REQ pfnClassReq, HSUSBD_SE
     /* Initial USB engine */
     HSUSBD->PHYCTL |= (HSUSBD_PHYCTL_PHYEN_Msk | HSUSBD_PHYCTL_DPPUEN_Msk);
     /* wait PHY clock ready */
+	
+		while((HSUSBD->BUSINTSTS & HSUSBD_BUSINTEN_PHYCLKVLDIEN_Msk) 		
+					!= HSUSBD_BUSINTEN_PHYCLKVLDIEN_Msk ){}
+							
     while (1)
     {
         HSUSBD->EP[EPA].EPMPS = 0x20ul;
```

stankong, 2020-04-09:
从数据表得知，HSUSBD_BUSINTSTS_PHYCLKVLDIF标志可以指示PHY clock ready ，改进补丁如下
`while((HSUSBD->BUSINTSTS & HSUSBD_BUSINTEN_PHYCLKVLDIEN_Msk)!= HSUSBD_BUSINTEN_PHYCLKVLDIEN_Msk ){}`
同时发现当USB仅供电的时候会在此处while死循环，最终修改 HSUSBD_Open函数如下，可以解决上述两个问题
```
void HSUSBD_Open(S_HSUSBD_INFO_T *param, HSUSBD_CLASS_REQ pfnClassReq, HSUSBD_SET_INTERFACE_REQ pfnSetInterface)
{
    g_hsusbd_sInfo = param;
    g_hsusbd_pfnClassRequest = pfnClassReq;
    g_hsusbd_pfnSetInterface = pfnSetInterface;

    /* get EP0 maximum packet size */
    g_hsusbd_CtrlMaxPktSize = g_hsusbd_sInfo->gu8DevDesc[7];

    /* Initial USB engine */
		HSUSBD_ENABLE_PHY();
    /* wait PHY clock ready */
							
//		while(1)
//		{
//				HSUSBD->BUSINTEN = (HSUSBD_BUSINTEN_RESUMEIEN_Msk|HSUSBD_BUSINTEN_SUSPENDIEN_Msk|HSUSBD_BUSINTEN_RSTIEN_Msk|HSUSBD_BUSINTEN_VBUSDETIEN_Msk);
//				if(HSUSBD->BUSINTEN == (HSUSBD_BUSINTEN_RESUMEIEN_Msk|HSUSBD_BUSINTEN_SUSPENDIEN_Msk|HSUSBD_BUSINTEN_RSTIEN_Msk|HSUSBD_BUSINTEN_VBUSDETIEN_Msk))
//					break;
//		}
							
		while((HSUSBD->BUSINTSTS & HSUSBD_BUSINTEN_PHYCLKVLDIEN_Msk) 		
						!= HSUSBD_BUSINTEN_PHYCLKVLDIEN_Msk ){}
										
		HSUSBD_CLR_SE0();
							
    while (1)
    {
        HSUSBD->EP[EPA].EPMPS = 0x20ul;
        if (HSUSBD->EP[EPA].EPMPS == 0x20ul)
        {
            break;
        }
    }
    /* Force SE0, and then clear it to connect*/
    HSUSBD_SET_SE0();
}
```

yachen, 2020-04-15:
Hi,

這樣的修改是正確的. 請問是否願意提 pull request 我來 merge, 或是我這邊直接修改?

Sincerely,

Yi-An Chen

stankong, 2020-04-15:
您可以直接修改。。。

发送自 Windows 10 版邮件应用

发件人: Yi-An Chen
发送时间: 2020年4月15日 17:41
收件人: OpenNuvoton/M480BSP
抄送: stankong; State change
主题: Re: [OpenNuvoton/M480BSP] 例程："HSUSBD_Mass_Storage_SD" 不能在192MHz主频下工作。 (#7)

Hi,
這樣的修改是正確的. 請問是否願意提 pull request 我來 merge, 或是我這邊直接修改?
Sincerely,
Yi-An Chen

yachen, 2020-04-15:
好的, 謝謝.