OpenNuvoton/M480BSP #9: LICENSE for the BSP/HAL

parthitce, 2020-04-23:
At @linumiz we are currently working on this M480 SoC port for Zephyr RTOS for one of our customer. As this port doesn't exist in Zephyr and HAL layer, we are in the phase of porting and analyzing it.

Zephyr RTOS and it's components are open source. So all it's future components should comply to the open source license forms. When adding M480BSP as separate HAL layer for Nuvoton, I couldn't find a clear information about the License in the source files and in this repository.

**Question:**

- What is the LICENSE for this M480 BSP/HAL layer?

- Will it be possible to add LICENSE file to this repository?

parthitce, 2020-04-26:
PR: https://github.com/zephyrproject-rtos/zephyr/pull/24704 is opened in Zephyr for m48x SoC support

Could you please give your feedback about License @yachen @cwweng-nuvoton @mjchen1 @ttnwosay ?

yachen, 2020-04-27:
Hi,

We're working on the license wording with legal team. 
In this license, we grant our user to obtain a copy of BSP free of charge and he can use, copy, modify, distribute or sell with the products using Nuvoton microcontrollers or microprocessors without limitation. Thanks.

Sincerely,

Yi-An Chen

parthitce, 2020-04-27:
> Hi,
>
> We're working on the license wording with legal team.
> In this license, we grant our user to obtain a copy of BSP free of charge and he can use, copy, modify, distribute or sell with the [...]

@yachen Thank you very much. As we need legally approved version, I will wait for your update on this.

parthitce, 2020-05-08:
@yachen https://github.com/zephyrproject-rtos/zephyr/issues/24703 is currently blocked and waiting for the License. Could you please share further update or rough estimation/timeline for the License availability?

ssekar15, 2020-05-11:
@walterjstseng We would like to get clarification about the License in order to proceed the support for  Nuvoton BSP on Zephyr RTOS. Would like to know plan/insight from you. Thanks in advance.

yachen, 2020-05-14:
@parthitce We add Apache-2.0 License to driver source code (HAL) in 08be9f19. Thanks.

parthitce, 2020-05-14:
@yachen Thank you very much. We will now integrate the repository in Zephyr and will close this issue once everything goes smooth.

closed 2020-05-28 by yachen