OpenNuvoton/M480BSP #16: M484 USB HS HOST example, with DMA

Hoel, 2020-08-29:
Hi,
We are working on the second target which is an M484SIDAE, in this use case the M484 is an HS USB HOST and it need to acquire datas continuously from an external USB HS chip, The M484 also has to perform intensive realtime signal processing on tthe acquired datas thus it is important that DMA is used as much as possible to handle USB data transfers. I have parsed the HSUSB examples of the M480 BSP and i cannot find an example of USB HS host with DMA, could you kindly point us to such example?

As prelimimary work we already implemented the driver for our chip in an STM32F411 (after porting libusb on this STM32), it works fine however the STM32F411 can only do USB FS and without DMA.

yachen, 2020-08-31:
Hi,

HSUSB (EHCI) contains its own DMA. Thus you cannot find HSUSB sample using PDMA to transfer USB data.
All HSUSBH sample codes utilize its own DMA to transfer data, you can develop your project base on a sample code implementing the same USB class as your system.

Sincerely,

Yi-An Chen

yachen, 2020-08-31:
USBH1.1 (OHCI) also contains its own DMA.  So no matter you're connection with USB1.1 or USB2.0 device, the host controller in M480 always use its own DMA to do the transfer.

Hoel, 2020-08-31:
Hi,
Thank you, i have written all the control transfer (ctrl_xfer) part of the driver already and it it is working fine, however i struggle with the endpoint and data-in pipe to setup the actual data acquisition

yachen, 2020-08-31:
Hi,

Are you using standard USB class (mass  storage, HID...)?

Sincerely,

Yi-An Chen

Hoel, 2020-08-31:
No, the device is the RTL2832U,  the chip offers 2 endpoints :  1 control pipe, and 1 data pipe for TS transfers
 the data pipe can be configured in ISO or BULK mode. On STM32 i used BULK, which seems to make sense since i only receive datas, anyway the goal is to have the largest transfers and least overhead (interrupts) so most processor time can go to signal processing in between transfers.

Hoel, 2020-08-31:
i started from the LBK demo and added my functions into lbk_test.c (was main.c) lbk_driver.c and lbk_driver.h
[User.zip](https://github.com/OpenNuvoton/M480BSP/files/5147805/User.zip)

Hoel, 2020-08-31:
As you can see the setup is alright
```
dev_write_reg SUCCESS! ret:0  len:1
dev_write_reg SUCCESS! ret:0  len:1
dev_write_reg SUCCESS! ret:0  len:2
dev_write_reg SUCCESS! ret:0  len:2
dev_write_reg SUCCESS! ret:0  len:1
dev_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:2
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
ev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:1
USB_EPA_MAXPKT 00020000
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:2
dev_demod_read_reg SUCCESS! ret:0  len:1
dev_demod_write_reg SUCCESS! ret:0  len:2
```

Hoel, 2020-08-31:
[image]
[image]
[image]
[image]
[image]

Hoel, 2020-08-31:
i tried to add the bulk transfer test function just after my setup, but sure enough it did not work:

```
    msg_tick = get_ticks();
    xfer_len = 512;
    for (loop = 1; loop<5; loop++)
    {
        if (!kbhit()){ if (getchar() == 'x') return; }
        if (lbk_bulk_read(buff_in, xfer_len, 100) != 0)
        {
            printf("Bulk-in transfer failed. Stop bulk transfer loop.\n");
            return;
        }
        printf("Bulk cnt %d\n",loop);
        if (get_ticks() - msg_tick >= 100)
        {
            printf("Bulk transfer loop count %d.    \r", loop);
            msg_tick = get_ticks();
        }
    }
```

```
Dummy write OK
USB_SYSCTL OK
USB_EPA_MAXPKT OK
USB_EPA_CTL OK
DEMOD_CTL_1 OK
DEMOD_CTL OK
DEMOD RST 1 OK
DEMOD RST 2 OK
SPECTRUM INVERSION OK
CHANNEL REJECTION OK
DDC shift / IF frequency OK
Set FIR OK
SDR mode OK
FSM reg OK
AGC disable OK
IF & RF AGC loop disable OK
PID filter disable OK
ADC_I/ADC_Q datapath OK
Zero-IF, DC cancellation, IQ est/comp OK
P_CK0 4MHz disable OK
Set sample rate OK
Test mode OK, USB_EPA_MAXPKT 00020000
Reset buffer 1 OK
Reset buffer 2 OK
Bulk transfer ready
qTD error token=0x2008d40!  0x20027dd8
Bulk cnt 1
Bulk-in transfer failed. Stop bulk transfmyDum write OK
USB_SYSCTL OK
USB_EPA_MAXPKT OK
USB_EPA_CTL OK
DEMOD_CTL_1 OK
DEMOD_CTL OK
DEMOD RST 1 OK
DEMOD RST 2 OK
SPECTRUM INVERSION OK
CHANNEL REJECTION OK
DDC shift / IF frequency OK
Set FIR OK
SDR mode OK
FSM reg OK
AGC disable OK
IF & RF AGC loop disable OK
PID filter disable OK
ADC_I/ADC_Q datapath OK
Zero-IF, DC cancellation, IQ est/comp OK
P_CK0 4MHz disable OK
Set sample rate OK
Test mode OK, USB_EPA_MAXPKT 00020000
Reset buffer 1 OK
Reset buffer 2 OK
Bulk transfer ready
qTD error token=0x2008d40!  0x20027dd8
Bulk cnt 1
Bulk-in transfer failed. Stop bulk transfer loop.
```

Hoel, 2020-08-31:
enable the memory debug, the bulk-in endoint is recognized:

```
+--------------------------------------+
|                                      |
|  USB Host Vendor Loopback Demo       |
|                                      |
+--------------------------------------+
[ALLOC] [ED] - 0x20001140
[ALLOC] [ED] - 0x20001180
[ALLOC] [ED] - 0x200011c0
[ALLOC] [ED] - 0x20001200
[ALLOC] [ED] - 0x20001240
[ALLOC] [ED] - 0x20001280
[ALLOC] [qTD] - 0x200012c0
[ALLOC] [QH] - 0x20001300
[ALLOC] [QH] - 0x20001340
[ALLOC] [QH] - 0x20001380
[ALLOC] [QH] - 0x200013c0
[ALLOC] [QH] - 0x20001400
[ALLOC] [QH] - 0x20001440
[ALLOC] [QH] - 0x20001480
[ALLOC] [QH] - 0x200014c0
[ALLOC] [QH] - 0x20001500
[ALLOC] [QH] - 0x20001540
[ALLOC] [QH] - 0x20001580
[ALLOC] [QH] - 0x200015c0
Connect device =>
[ALLOC] [UTR] - 0x20005820
[ALLOC] [QH] - 0x20001600
[ALLOC] [qTD] - 0x20001640
[ALLOC] [qTD] - 0x20001680
[ALLOC] [qTD] - 0x200016c0
[FREE] [UTR] - 0x2[FREE]  [qTD] - 0x200016c0
[FREE]  [qTD] - 0x20001680
[FREE]  [qTD] - 0x20001640
20001640
[ALLOC] [UTR] - 0x20005820
[ALLOC] [qTD] - 0x20001700
[ALLOC] [qTD] - 0x20001740
[FREE] [UTR] - 0x2[FREE]  [qTD] - 0x20001740
[FREE]  [qTD] - 0x20001700
20001700
New high-speed device address 1 assigned.
[ALLOC] [UTR] - 0x20005820
[ALLOC] [qTD] - 0x20001780
[ALLOC] [qTD] - 0x200017c0
[ALLOC] [qTD] - 0x20001800
[FREE] [UTR] - 0x2[FREE]  [qTD] - 0x20001800
[FREE]  [qTD] - 0x200017c0
[FREE]  [qTD] - 0x20001780
20001780
[ALLOC] [UTR] - 0x20005820
[ALLOC] [qTD] - 0x20001840
[ALLOC] [qTD] - 0x20001880
[ALLOC] [qTD] - 0x200018c0
[FREE] [UTR] - 0x2[FREE]  [qTD] - 0x200018c0
[FREE]  [qTD] - 0x20001880
[FREE]  [qTD] - 0x20001840
20001840
[ALLOC] [UTR] - 0x20005820
[ALLOC] [qTD] - 0x20001900
[ALLOC] [qTD] - 0x20001940
[ALLOC] [qTD] - 0x20001980
[FREE] [UTR] - 0x2[FREE]  [qTD] - 0x20001980
[FREE]  [qTD] - 0x20001940
[FREE]  [qTD] - 0x20001900
20001900
[ALLOC] [UTR] - 0x20005820
[ALLOC] [qTD] - 0x200019c0
[ALLOC] [qTD] - 0x20001a00
[FREE] [UTR] - 0x2[FREE]  [qTD] - 0x20001a00
[FREE]  [qTD] - 0x200019c0
200019c0
lbk_probe - device found (vid=0xbda, pid=0x2838), interface 0.
Device endpoints : 1
Bulk-in endpoint: 0x81
A Vendor LBK device is connected, do not support multiple devices!
Device [0xbda,0x2838] was connected.
Speed:    high-speed
Location: USB 2.0 port
```

yachen, 2020-08-31:
Are you connecting with RTL2832U or M480 USB device itself?
The sample doesn't support USB Host connect to USB device on the same device.

Hoel, 2020-08-31:
of course, the RTL2832U is connected to M480, i know that the sample is for two M480 but i modified it to work for my use case, and as you can see i can configure the chip correctly, the only problem is bulk transfer.

ychuang3, 2020-08-31:
Hi,

Looks like the USB device enumeration is fine. But only the first interface (0) is recognized, and the second interface failed.
- interface 0:   "lbk_probe - device found (vid=0xbda, pid=0x2838), interface 0."
- interface ?:   "A Vendor LBK device is connected, do not support multiple devices!"  The probe of second interface failed and return from here.   You have to modify the driver to support multiple interfaces. Add some code to distinguish USB interface # and implement corresponding functions for these interfaces.  Note that he lbk sample has only one interface.

Please use  "ret = lbk_bulk_read(buff_in, xfer_len, 100);"  and print out the ret value. It will be helpful to debug.
The bulk transfer length of lbk device is always 512 bytes, please check the bulk-in data length of RTL2832U.
By the way, in lbk_bulk_read() you can  get the actually received byte count of lbk_bulk_read() by utr->xfer_len once usbh_bulk_xfer() success.

lbk_bulk_read() is in blocking style. You can also modify it as non-blocking style by install a callback function to utr->func. The callback function will be called when the bulk transfer was completed or failed. Thus you can return from lbk_bulk_read() after invoking usbh_bulk_xfer(), instead of waiting for "utr->bIsTransferDone" flag. That will be more efficient.

 Y.C. Huang

Hoel, 2020-08-31:
Ah, thank you for the good suggestions, i also found weird that only 1 endpoint was recognized, so it seems that so far i sent the Control Transfer Commands to the data bulk-in endpoint, right?

> The bulk transfer length of lbk device is always 512 bytes, please check the bulk-in data length of RTL2832U.
Right, by default USB_EPA_MAXPKT is set to 64 (for FS mode, which i do not use) and i set it to 0x200 (512, HS mode), according to datasheet it can go up to 1023 in HS mode and 1024 in FS mode.

> Please use "ret = lbk_bulk_read(buff_in, xfer_len, 100);" and print out the ret value. It will be helpful to debug.
OK i will check the return value.

> lbk_bulk_read() is in blocking style. You can also modify it as non-blocking style by install a callback function to utr->func
Right, that was my second immediate concern, of course non blocking is mandatory in my case, otherwise there is no way to process the data in between transfers.

Hoel, 2020-08-31:
i just recall, i previously added a debug statement in lbk_probe to see how many endpoints were present, and it only found 1, what is the reason for this? obviously there is actually 2 endpoint, do i need to modify something else in the lbk_driver so the two endpoints get recognized?

Hoel, 2020-08-31:
ah, i just commented :
`return USBH_ERR_NOT_MATCHED;`
and now it can see both interfaces:

```
Connect device =>
New high-speed device address 1 assigned.
lbk_probe - device found (vid=0xbda, pid=0x2838), interface 0.
Device endpoints nbr: 1
Bulk-in endpoint: 0x81

lbk_probe - device found (vid=0xbda, pid=0x2838), interface 1.
Device endpoints nbr: 0
Device [0xbda,0x2838] was connected.
Speed:    high-speed
Location: USB 2.0 port
```

Hoel, 2020-08-31:
so here is the current status:

the probe function can see two interfaces, but the second interface shows 0 endpoint, either in interface descriptor or alternative interface descriptor:

```
Connect device =>
New high-speed device address 1 assigned.
lbk_probe - device found (vid=0xbda, pid=0x2838), interface 0.
Device endpoints nbr: 1
ifd->bNumEndpoints : 1

Bulk-in endpoint: 0x81

lbk_probe - device found (vid=0xbda, pid=0x2838), interface 1.
Device endpoints nbr: 0
ifd->bNumEndpoints : 0

Device [0xbda,0x2838] was connected.
Speed:    high-speed
Location: USB 2.0 port
```

The bulk transfer fails with -203

```
qTD error token=0x2008d40!  0x20027de0
Bulk cnt 1
Bulk-in transfer failed. ret=-203
```

ychuang3, 2020-08-31:
Hi,

One interface can be a logical device. For example, a camera may have a video class interface and a mass storage class interface. Another example, M480 BSP USBD_VCOM_DualPort sample is a dual VCOM port device, which has two similar interfaces each for a VCOM port. Both interfaces use the same VCOM driver. So, the VCOM driver probe function will be called twice.
 
All interfaces of the same USB device use the same endpoint 0 for control transfer. The probe function report the total number of endpoints in the interface, but exclude control endpoint (endpoint # 0). 

Please enable "DUMP_DESCRIPTOR" in bsp\Library\UsbHostLib\inc\config.h. This can dump configuration descriptor of USB device. We can know the detail of RTL2832U from here.

Y.C. Huang

Hoel, 2020-08-31:
```
Connect device =>
New high-speed device address 1 assigned.

[Device Descriptor]
----------------------------------------------
  Length              = 18
  DescriptorType      = 0x01
  USB version         = 2.00
  Vendor:Product      = 0bda:2838
  MaxPacketSize0      = 64
  NumConfigurations   = 1

  Device version      = 1.00
  Device Class:SubClass:Protocol = 00:00:00

[Configuration Descriptor]
----------------------------------------------
  Length              =  9
  DescriptorType      = 02
  wTotalLength        = 34
  bNumInterfaces      = 2

  bConfigurationValue = 1

  iConfiguration      = 4

  bmAttributes        = 0x80
  MaxPower            = 250

    [Interface Descriptor]

    ----------------------------------------------
      Length              =  9
      DescriptorType      = 04
      bInterfaceNumber    = 0

      bAlternateSetting   = 0

      bNumEndpoints       = 1

      bInterfaceClass     = 0xff
      bInterfaceSubClass  = 0xff
      bInterfaceProtocol  = 0xff
      iInterface          = 5

        [Endpoint Descriptor]
        ----------------------------------------------
          Length              =  7
          DescriptorType      = 05
          bEndpointAddress    = 0x81
          bmAttributes        = 0x02
          wMaxPacketSize      = 512

          bInterval           = 0
          bRefresh            = 9
          bSynchAddress       = 4

    [Interface Descriptor]
    ----------------------------------------------
      Length              =  9
      DescriptorType      = 04
      bInterfaceNumber    = 1

      bAlternateSetting   = 0

      bNumEndpoints       = 0

      bInterfaceClass     = 0xff
      bInterfaceSubClass  = 0xff
      bInterfaceProtocol  = 0xff
      iInterface          = 5
Parsing CONFIG =>
Descriptor Found - Interface 0, Alt: 0, num_alt:0, remaining len: 16
Finding 1 endpoints of interface 0, alt 0...
Descriptor Found - Alt: 0, Endpoint 0x81, remaining len: 16
EP parse remaining 9
lbk_probe - device found (vid=0xbda, pid=0x2838), interface 0.
Device endpoints nbr: 1
ifd->bNumEndpoints : 1

ifd->iInterface : 5
Bulk-in endpoint: 0x81

IFACE parse remaining 9
Descriptor Found - Interface 1, Alt: 0, num_alt:0, remaining len: 0

lbk_probe - device found (vid=0xbda, pid=0x2838), interface 1.
Device endpoints nbr: 0
ifd->bNumEndpoints : 0

ifd->iInterface : 5
IFACE parse remaining 0
Device [0xbda,0x2838] was connected.
Speed:    high-speed
Location: USB 2.0 port
```

ychuang3, 2020-08-31:
Hi,

So the device has only one bulk-in endpoint in interface 0. Interface 1 use control transfer only.

The error code -203 is time-out error (refer to usbh_lib.h). You have set the time-out as 100 ms. It can be:
1. RTL2832U is still not ready in 100ms. (100ms is enough for lbk device, but may not for RTL2832U) 
2. RTL2832U has no data for Host. Maybe not in work state.

It cannot be other errors, because USB Host hardware did not report errors. The  "qTD error token=0x2008d40!" is caused by driver force to terminate transfer (because of time-out).

For debug, suggest try to give time-out 5 seconds , lbk_bulk_read(buff_in, xfer_len, 5000). If it still get time-out error, you can determine that RTL2832U has no data for Host. You may have to issue some control messages to make RTL2832U work.

Y.C. Huang

Hoel, 2020-08-31:
ah, i see, thank you for the insight, i will set the timeout to 5000mS however it is sure that the device is ready, even well before 100mS, so there must be something wrong in my setup, even tho no error is returned for all commands.
Digging on that.

Hoel, 2020-08-31:
i increased the timeout to 5000ms, still get the "qTD error token=0x2008d40!" error, however the return code is now 0, strange...

ychuang3, 2020-09-01:
Hi, 

You should have called usbh_ctrl_xfer() to do the control transfers. It return 0 only when control transfer was sent and ACK by device successfully.

I think the driver may not select the correct interface. At the end of lbk_probe(), it assign the interface to lbk  "g_lbk_dev.iface = iface;".  Driver probe RTL2832U  interface 0 and assign to  "g_lbk_dev.iface", but it assign interface 1 again in the next probe of interface 1.
You should add  "if (ifd->bInterfaceNumber != 0) return 0;"  before that.

Please modify driver and keep timeout 5 seconds.

Y.C.Huang

ychuang3, 2020-09-01:
Hi,

In usbh_ctrl_xfer() of usb_core.c, please dump the setup packet (utr->setup). Also dump  STM32 issued setup packets. Check raw data packets can make sure you have do the same thing with STM32. 

Y.C. Huang

Hoel, 2020-09-01:
thank you very much, indeed i suspect the driver is using the wrong interface, that said yesterday i check that the endpoint used by the driver was well 0x81, and it was.i am going to try your suggestions now and see what the outcome is.

Hoel, 2020-09-01:
> You should have called usbh_ctrl_xfer() to do the control transfers. It return 0 only when control transfer was sent and ACK by device successfully.
> 

That is what i used for control transfers indeed

```
int dev_write_reg(uint8_t block,uint16_t addr,uint16_t val, uint8_t len)
{
    uint32_t   xfer_len = 0;
    int        ret;

    if (g_lbk_dev.udev == NULL) return -1;

	// "index"
    uint16_t regWriteIndex;
	regWriteIndex = (block << 8) | 0x10;

	// "data"
	uint8_t regWriteData[2];
	if (len == 1) regWriteData[0] = val & 0xff;
	else regWriteData[0] = val >> 8;
	regWriteData[1] = val & 0xff;

    ret = usbh_ctrl_xfer(g_lbk_dev.udev,
                         REQ_TYPE_OUT | REQ_TYPE_VENDOR_DEV | REQ_TYPE_TO_DEV,  /* bmRequestType : vendor, host to device, device  */
                         0/*REQ_SET_DATA*/,                  /* bRequest*/
						 addr,                          /* wValue */
						 regWriteIndex,                 /* wIndex */
                         len,                           /* wLength */
						 regWriteData,                  /* data buffer */
                         &xfer_len,                     /* actual length of data send */
                         100);

    if ((ret < 0) || (xfer_len != len))
    {
        printf("dev_write_reg failed! %d, %d\n", ret, xfer_len);
        return ret;
    }
    //printf("dev_write_reg SUCCESS! ret:%d  len:%d\n", ret, xfer_len);
    return 0;
}
```

Hoel, 2020-09-01:
added the statement to avoid assignation of interface 1, still same problem : 

```
Connect device =>
New high-speed device address 1 assigned.
[Device Descriptor]
----------------------------------------------
  Length              = 18
  DescriptorType      = 0x01
  USB version         = 2.00
  Vendor:Product      = 0bda:2838
  MaxPacketSize0      = 64
  NumConfigurations   = 1
  Device version      = 1.00
  Device Class:SubClass:Protocol = 00:00:00
[Configuration Descriptor]
----------------------------------------------
  Length              =  9
  DescriptorType      = 02
  wTotalLength        = 34
  bNumInterfaces      = 2
  bConfigurationValue = 1
  iConfiguration      = 4
  bmAttributes        = 0x80
  MaxPower            = 250
    [Interface Descriptor]
    ----------------------------------------------
      Length              =  9
      DescriptorType      = 04
      bInterfaceNumber    = 0
      bAlternateSetting   = 0
      bNumEndpoints       = 1
      bInterfaceClass     = 0xff
      bInterfaceSubClass  = 0xff
      bInterfaceProtocol  = 0xff
      iInterface          = 5
        [Endpoint Descriptor]
        ----------------------------------------------
          Length              =  7
          DescriptorType      = 05
          bEndpointAddress    = 0x81
          bmAttributes        = 0x02
          wMaxPacketSize      = 512
          bInterval           = 0
          bRefresh            = 9
          bSynchAddress       = 4
    [Interface Descriptor]
    ----------------------------------------------
      Length              =  9
      DescriptorType      = 04
      bInterfaceNumber    = 1

      bAlternateSetting   = 0
      bNumEndpoints       = 0
      bInterfaceClass     = 0xff
      bInterfaceSubClass  = 0xff
      bInterfaceProtocol  = 0xff
      iInterface          = 5
Parsing CONFIG =>
Descriptor Found - Interface 0, Alt: 0, num_alt:0, remaining len: 16
Finding 1 endpoints of interface 0, alt 0...
Descriptor Found - Alt: 0, Endpoint 0x81, remaining len: 16
EP parse remaining 9
lbk_probe - device found (vid=0xbda, pid=0x2838), interface 0.
Device endpoints nbr: 1
ifd->bNumEndpoints : 1

ifd->iInterface : 5
Bulk-in endpoint: 0x81
IFACE parse remaining 9
Descriptor Found - Interface 1, Alt: 0, num_alt:0, remaining len: 0
lbk_probe - device found (vid=0xbda, pid=0x2838), interface 1.
Device endpoints nbr: 0
ifd->bNumEndpoints : 0
ifd->iInterface : 5
IFACE parse remaining 0
Device [0xbda,0x2838] was connected.
Speed:    high-speed
Location: USB 2.0 port
Dummy write OK
USB_SYSCTL OK
USB_EPA_MAXPKT OK
USB_EPA_CTL OK
DEMOD_CTL_1 OK
DEMOD_CTL OK
DEMOD RST 1 OK
DEMOD RST 2 OK
SPECTRUM INVERSION OK
CHANNEL REJECTION OK
DDC shift / IF frequency OK
Set FIR OK
SDR mode OK
FSM reg OK
AGC disable OK
IF & RF AGC loop disable OK
PID filter disable OK
ADC_I/ADC_Q datapath OK
Zero-IF, DC cancellation, IQ est/comp OK
TP_CK0 4MHz disable OK
Set sample rate OK
USB_EPA_MAXPKT = 0x00020000
Reset buffer 1 OK
Reset buffer 2 OK
Bulk transfer ready
Bulk-in transfer from endpoint : 0x81
qTD error token=0x2008d40!  0x20027de8
transfer cnt 1
```

Hoel, 2020-09-01:
```
+--------------------------------------+
|                                      |
|  USB Host Vendor Loopback Demo       |
|                                      |
+--------------------------------------+
Connect device =>

[setup packet]
----------------------------------------------
  bmRequestType       = 0x80
  bRequest            = 0x06
  wValue              = 256
  wIndex              =  0
  wLength             = 18

[setup packet]
----------------------------------------------
  bmRequestType       = 0x00
  bRequest            = 0x05
  wValue              =  1
  wIndex              =  0
  wLength             =  0
New high-speed device address 1 assigned.

[setup packet]
----------------------------------------------
  bmRequestType       = 0x80
  bRequest            = 0x06
  wValue              = 256
  wIndex              =  0
  wLength             = 18

[Device Descriptor]
----------------------------------------------
  Length              = 18
  DescriptorType      = 0x01
  USB version         = 2.00
  Vendor:Product      = 0bda:2838
  MaxPacketSize0      = 64
  NumConfigurations   = 1
  Device version      = 1.00
  Device Class:SubClass:Protocol = 00:00:00

[setup packet]
----------------------------------------------
  bmRequestType       = 0x80
  bRequest            = 0x06
  wValue              = 512
  wIndex              =  0
  wLength             =  9

[setup packet]
----------------------------------------------
  bmRequestType       = 0x80
  bRequest            = 0x06
  wValue              = 512
  wIndex              =  0
  wLength             = 34

[Configuration Descriptor]
----------------------------------------------
  Length              =  9
  DescriptorType      = 02
  wTotalLength        = 34
  bNumInterfaces      = 2
  bConfigurationValue = 1
  iConfiguration      = 4
  bmAttributes        = 0x80
  MaxPower            = 250
[... 779 lines cut ...]
  wIndex              = 10
  wLength             =  1
Reset buffer 1 OK

[setup packet]
----------------------------------------------
  bmRequestType       = 0x40
  bRequest            = 0x00
  wValue              = 18464
  wIndex              = 17
  wLength             =  2

[setup packet]
----------------------------------------------
  bmRequestType       = 0xc0
  bRequest            = 0x00
  wValue              = 288
  wIndex              = 10
  wLength             =  1
Reset buffer 2 OK
Bulk transfer ready
Bulk-in transfer from endpoint : 0x81
qTD error token=0x2008d40!  0x20027de8
transfer cnt 1
```

ychuang3, 2020-09-01:
Hi,

"USB_EPA_MAXPKT  is 0x00020000" seems be problematic.
Refer to Linux driver, the data payload of write register 0x2158 should be "\x00\x02\x00\x00", i.e. 0x00000200 in little endian word.

As you have a workable platform, i suggest to dump and compare  the raw data of setup packets. 

Y.C. Huang

Hoel, 2020-09-01:
i see, but we write 0x0002 in the first two registers only
i am in the process of making comparative registers dump
below is the setup with STM32, with success bulk  and M484, with fail bulk

```
System ready 
Starting USB
USB PHY init 
USB Device Connected
USB Device Reset Completed
DEBUG : USBH_UserProcess
DEBUG : HOST_USER_CONNECTION
PID: 2838h
VID: bdah
Address (#1) assigned.
Manufacturer : Realtek
Product : RTL2838UHIDIR
Serial Number : 00000001
Enumeration done.
This device has only 1 configuration.
Default configuration set.
--> pipe allocated
--> pipe ID : 2
finished baseband init
rtlsdr_set_sample_rate
Sampling at 240000 S/s.
Device init done
Starting data capture
91.70 kBps
469.32 kBps
468.02 kBps
469.86 kBps
468.45 kBps
```
```
Parsing CONFIG =>
Descriptor Found - Interface 0, Alt: 0, num_alt:0, remaining len: 16
Finding 1 endpoints of interface 0, alt 0...
Descriptor Found - Alt: 0, Endpoint 0x81, remaining len: 16
EP parse remaining 9
lbk_probe - device found (vid=0xbda, pid=0x2838), interface 0.
Device endpoints nbr: 1
ifd->bNumEndpoints : 1
ifd->iInterface : 5
Bulk-in endpoint: 0x81
IFACE parse remaining 9
Descriptor Found - Interface 1, Alt: 0, num_alt:0, remaining len: 0
lbk_probe - device found (vid=0xbda, pid=0x2838), interface 1.
Device endpoints nbr: 0
ifd->bNumEndpoints : 0
ifd->iInterface : 5
IFACE parse remaining 0
Device [0xbda,0x2838] was connected.
Speed:    high-speed
Location: USB 2.0 port

finished baseband init
rtlsdr_set_sample_rate
Samplerate OK 0
Sampling at 240000 S/s.
Device init done
Starting data capture
Bulk-in transfer from endpoint : 0x81
Bulk-in transfer failed. ret=-203
```

I will send the same with actual register values in a very short while

Hoel, 2020-09-01:
> "USB_EPA_MAXPKT is 0x00020000" seems be problematic.
> Refer to Linux driver, the data payload of write register 0x2158 should be "\x00\x02\x00\x00", i.e. 0x00000200 in little endian word.

Yes i noticed that, however that is how its done in working driver, we write 0x0002 to the first to bytes of the register. On theS STM32 this call is disabled to let the 64byte default size, do you think i can try 64 byte buffer on M484 in HS mode? I tried it yesterday but bulk failed too

ychuang3, 2020-09-01:
[setup packet]
----------------------------------------------
  bmRequestType       = 0xc0
  bRequest            = 0x12
  wValue              = 0x2158
  wIndex              = 256
  wLength             =  4
USB_EPA_MAXPKT = 0x00020000

int usbh_ctrl_xfer(UDEV_T *udev, uint8_t bmRequestType, uint8_t bRequest, uint16_t wValue, uint16_t wIndex,
                   uint16_t wLength, uint8_t *buff, uint32_t *xfer_len, uint32_t timeout)

the content of <buff> should be  "\x00\x02\x00\x00"

You have assigned wLength 4, the driver will send 4 bytes to write register USB_EPA_MAXPKT.

Hoel, 2020-09-01:
i tried two things : i wrote a function to write 4 bytes insteqd of 2:

```
int dev_write_regs(uint8_t block,uint16_t addr,uint32_t val, uint8_t len)
{
    uint32_t   xfer_len = 0;
    int        ret;

    if (g_lbk_dev.udev == NULL) return -1;

	// "index"
    uint16_t regWriteIndex;
	regWriteIndex = (block << 8) | 0x10;

	// "data"
	uint8_t regWriteData[4];
	regWriteData[0] = ((val >> 24) & 0xff);
    regWriteData[1] = ((val >> 16) & 0xff);
    regWriteData[2] = ((val >> 8) & 0xff);
    regWriteData[3] = (val & 0xff);

    ret = usbh_ctrl_xfer(g_lbk_dev.udev,
                         REQ_TYPE_OUT | REQ_TYPE_VENDOR_DEV | REQ_TYPE_TO_DEV,  /* bmRequestType : vendor, host to device, device  */
                         0/*REQ_SET_DATA*/,                  /* bRequest*/
						 addr,                          /* wValue */
						 regWriteIndex,                 /* wIndex */
                         len,                           /* wLength */
						 regWriteData,                  /* data buffer */
                         &xfer_len,                     /* actual length of data send */
                         100);

    if ((ret < 0) || (xfer_len != len))
    {
        printf("dev_write_reg failed! %d, %d\n", ret, xfer_len);
        return ret;
    }
    //printf("dev_write_reg SUCCESS! ret:%d  len:%d\n", ret, xfer_len);
    return 0;
}
```

and wrote 0x00000200
it did not worked, the readout gave 0x00000000
`USB_EPA_MAXPKT = 0x00000000`

Then i used the normal function which write two bytes, and wrote 512 (so not in hex format) its still wrong: 
`USB_EPA_MAXPKT = 0x02000000`

```
int dev_write_reg(uint8_t block,uint16_t addr,uint16_t val, uint8_t len)
{
    uint32_t   xfer_len = 0;
    int        ret;

    if (g_lbk_dev.udev == NULL) return -1;

	// "index"
    uint16_t regWriteIndex;
	regWriteIndex = (block << 8) | 0x10;

	// "data"
	uint8_t regWriteData[2];
	if (len == 1) regWriteData[0] = val & 0xff;
	else regWriteData[0] = val >> 8;
	regWriteData[1] = val & 0xff;

    ret = usbh_ctrl_xfer(g_lbk_dev.udev,
                         REQ_TYPE_OUT | REQ_TYPE_VENDOR_DEV | REQ_TYPE_TO_DEV,  /* bmRequestType : vendor, host to device, device  */
                         0/*REQ_SET_DATA*/,                  /* bRequest*/
						 addr,                          /* wValue */
						 regWriteIndex,                 /* wIndex */
                         len,                           /* wLength */
						 regWriteData,                  /* data buffer */
                         &xfer_len,                     /* actual length of data send */
                         100);

    if ((ret < 0) || (xfer_len != len))
    {
        printf("dev_write_reg failed! %d, %d\n", ret, xfer_len);
        return ret;
    }
    //printf("dev_write_reg SUCCESS! ret:%d  len:%d\n", ret, xfer_len);
    return 0;
}
```

Hoel, 2020-09-01:
here is the call and the function used to  readback the register : 

```
	// Check EPA max pkt size
	if (dev_read_array( USBB,USB_EPA_MAXPKT,bfr,4) != 0){ printf("test mode failed\n");  return; }
	printf("USB_EPA_MAXPKT = 0x%02x%02x%02x%02x\n",bfr[0],bfr[1],bfr[2],bfr[3]);
```

```
int dev_read_array( uint8_t block, uint16_t addr,  uint8_t *array, uint8_t len){

	//dev_HandleTypeDef *dev_Handle = (dev_HandleTypeDef*) phost->pActiveClass->pData;
	int ret;
	uint32_t   xfer_len = 0;

	// "index"
	uint16_t arrReadIndex = (block << 8) ;

	ret = usbh_ctrl_xfer(g_lbk_dev.udev,
						 REQ_TYPE_IN | REQ_TYPE_VENDOR_DEV | REQ_TYPE_TO_DEV,  /* bmRequestType : vendor, device to host, to device  */
						 REQ_GET_DATA,                  /* bRequest*/
						 addr,                          /* wValue */
						 arrReadIndex,                  /* wIndex */
						 len,                           /* wLength */
						 array,                			/* data buffer */
						 &xfer_len,                     /* data received length */
						 100);

	if ((ret < 0) || (xfer_len != len))
	{
		printf("dev_demod_read_reg failed! %d, %d\n", ret, xfer_len);
		return ret;
	}
	//printf("dev_read_array SUCCESS! ret:%d  len:%d\n", ret, xfer_len);
	return 0;
}
```

ychuang3, 2020-09-01:
If val is 512, the data endian is wrong
uint8_t regWriteData[4];
	regWriteData[0] = ((val >> 24) & 0xff);
    regWriteData[1] = ((val >> 16) & 0xff);
    regWriteData[2] = ((val >> 8) & 0xff);
    regWriteData[3] = (val & 0xff);
==>   "\x00\x00\x02\x00"  ==> 0x00020000 [ X ]

the correct data sequence should be 
    regWriteData[3] = ((val >> 24) & 0xff);
    regWriteData[2] = ((val >> 16) & 0xff);
    regWriteData[1] = ((val >> 8) & 0xff);
    regWriteData[0] = (val & 0xff);
==> "\x00\x02\x00\x00"  ==> 0x00000200  [ O ]

Hoel, 2020-09-01:
i tried various things and i cannot write this register to 0x00000200, all i can do is 0x02000000 or 0x00020000 or 0x00000000

Hoel, 2020-09-01:
```
int dev_write_regs(uint8_t block,uint16_t addr,uint32_t val, uint8_t len)
{
    uint32_t   xfer_len = 0;
    int        ret;

    if (g_lbk_dev.udev == NULL) return -1;

	// "index"
    uint16_t regWriteIndex;
	regWriteIndex = (block << 8) | 0x10;

	// "data"
	uint8_t regWriteData[4];
	regWriteData[3] = ((val >> 24) & 0xff);
	regWriteData[2] = ((val >> 16) & 0xff);
	regWriteData[1] = ((val >> 8) & 0xff);
	regWriteData[0] = (val & 0xff);

    ret = usbh_ctrl_xfer(g_lbk_dev.udev,
                         REQ_TYPE_OUT | REQ_TYPE_VENDOR_DEV | REQ_TYPE_TO_DEV,  /* bmRequestType : vendor, host to device, device  */
                         0/*REQ_SET_DATA*/,                  /* bRequest*/
						 addr,                          /* wValue */
						 regWriteIndex,                 /* wIndex */
                         len,                           /* wLength */
						 regWriteData,                  /* data buffer */
                         &xfer_len,                     /* actual length of data send */
                         100);

    if ((ret < 0) || (xfer_len != len))
    {
        printf("dev_write_reg failed! %d, %d\n", ret, xfer_len);
        return ret;
    }
    //printf("dev_write_reg SUCCESS! ret:%d  len:%d\n", ret, xfer_len);
    return 0;
}
```

```
	if (dev_write_regs(USBB, USB_EPA_MAXPKT, 512, 4) != 0){ printf("USB_EPA_MAXPKT failed\n"); return; }
	printf("USB_EPA_MAXPKT OK\n");
```

`USB_EPA_MAXPKT = 0x00020000`

Hoel, 2020-09-01:
so i tried that, same result aat the intial call, please not no bitwise operation is done on the readout, the packets are just displayed in the order they have been readback as is, so i guess 0x00 0x02 0x00 0x00 is correct

ychuang3, 2020-09-01:
Yes, the readback data of USB_EPA_MAXPKT register is correct.

Hoel, 2020-09-01:
so, i dont know, it seems that writing 0x002 on two bytes is correct, that is what each and every driver i checked does, the problem comes from elsewhere

Hoel, 2020-09-01:
also, the chip is warm, which is a sign it is initialized correctly

Hoel, 2020-09-01:
comparative dumps

M484: 

```
Device [0xbda,0x2838] was connected.
Speed:    high-speed
Location: USB 2.0 port

value=0x2000 data=0x0909
value=0x2000 data=0x0909
value=0x2148 data=0x1002
value=0x300b data=0x2222
value=0x3000 data=0xe8e8
value=0x0120 data=0x1414
value=0x0120 data=0x0000
value=0x0120 data=0x1010
value=0x0120 data=0x0000
value=0x1520 data=0x0000
value=0x0120 data=0x0000
value=0x1620 data=0x0000
value=0x0120 data=0x0000
value=0x1620 data=0x0000
value=0x0120 data=0x0000
value=0x1720 data=0x0000
value=0x0120 data=0x0000
value=0x1820 data=0x0000
value=0x0120 data=0x0000
value=0x1920 data=0x0000
value=0x0120 data=0x0000
value=0x1a20 data=0x0000
value=0x0120 data=0x0000
value=0x1b20 data=0x0000
value=0x0120 data=0x0000
value=0x1c20 data=0xcaca
value=0x0120 data=0x0100
value=0x1d20 data=0xdcdc
value=0x0120 data=0x0100
value=0x1e20 data=0xd7d7
value=0x0120 data=0x0100
value=0x1f20 data=0xd8d8
value=0x0120 data=0x0100
value=0x2020 data=0xe0e0
value=0x0120 data=0x0100
value=0x2120 data=0xf2f2
value=0x0120 data=0x0100
value=0x2220 data=0x0e0e
value=0x0120 data=0x0100
value=0x2320 data=0x3535
value=0x0120 data=0x0100
value=0x2420 data=0x0606
value=0x0120 data=0x0100
value=0x2520 data=0x5050
value=0x0120 data=0x0100
value=0x2620 data=0x9c9c
value=0x0120 data=0x0100
value=0x2720 data=0x0d0d
value=0x0120 data=0x0100
value=0x2820 data=0x7171
value=0x0120 data=0x0100
value=0x2920 data=0x1111
value=0x0120 data=0x0100
value=0x2a20 data=0x1414
value=0x0120 data=0x0100
value=0x2b20 data=0x7171
value=0x0120 data=0x0100
value=0x2c20 data=0x7474
value=0x0120 data=0x0100
value=0x2d20 data=0x1919
value=0x0120 data=0x0100
value=0x2e20 data=0x4141
value=0x0120 data=0x0100
value=0x2f20 data=0xa5a5
value=0x0120 data=0x0100
value=0x1920 data=0x0505
value=0x0120 data=0x0000
value=0x9320 data=0xf0f0
value=0x0120 data=0x0000
value=0x9420 data=0x0f0f
value=0x0120 data=0x0000
value=0x1120 data=0x0000
value=0x0120 data=0x0000
value=0x0420 data=0x0000
value=0x0120 data=0x0000
value=0x6120 data=0x6060
value=0x0120 data=0x0000
value=0x0620 data=0x8080
value=0x0120 data=0x0000
value=0xb120 data=0x1b1b
value=0x0120 data=0x0000
value=0x0d20 data=0x8383
value=0x0120 data=0x0000
finished baseband init
rtlsdr_set_sample_rate
value=0x9f20 data=0x0e00
value=0x0120 data=0x004c
value=0xa120 data=0x0000
value=0x0120 data=0x004c
value=0x3f20 data=0x0000
value=0x0120 data=0x004c
value=0x3e20 data=0x0000
value=0x0120 data=0x0000
value=0x0120 data=0x1414
value=0x0120 data=0x004c
value=0x0120 data=0x1010
value=0x0120 data=0x004c
Samplerate OK 0
Sampling at 240000 S/s.
Device init done
Starting data capture
value=0x2148 data=0x1002
value=0x2148 data=0x0000
Bulk-in transfer from endpoint : 0x81
Bulk-in transfer failed. ret=-203
```

```
System ready 
Starting USB
USB PHY init 
USB Device Connected
USB Device Reset Completed
DEBUG : USBH_UserProcess
DEBUG : HOST_USER_CONNECTION
PID: 2838h
VID: bdah
Address (#1) assigned.
Manufacturer : Realtek
Product : RTL2838UHIDIR
Serial Number : 00000001
Enumeration done.
This device has only 1 configuration.
Default configuration set.
--> pipe allocated
--> pipe ID : 2
value=0x2000 data=0x0909
value=0x2000 data=0x0909
value=0x2148 data=0x1002
value=0x300b data=0x2222
value=0x3000 data=0xe8e8
value=0x0120 data=0x1414
value=0x0120 data=0x0000
value=0x0120 data=0x1010
value=0x0120 data=0x0000
value=0x1520 data=0x0000
value=0x0120 data=0x0000
value=0x1620 data=0x0000
value=0x0120 data=0x0000
value=0x1620 data=0x0000
value=0x0120 data=0x0000
value=0x1720 data=0x0000
value=0x0120 data=0x0000
value=0x1820 data=0x0000
value=0x0120 data=0x0000
value=0x1920 data=0x0000
value=0x0120 data=0x0000
value=0x1a20 data=0x0000
value=0x0120 data=0x0000
value=0x1b20 data=0x0000
value=0x0120 data=0x0000
value=0x1c20 data=0xcaca
value=0x0120 data=0x0000
value=0x1d20 data=0xdcdc
value=0x0120 data=0x0000
value=0x1e20 data=0xd7d7
value=0x0120 data=0x0000
value=0x1f20 data=0xd8d8
value=0x0120 data=0x0000
value=0x2020 data=0xe0e0
value=0x0120 data=0x0000
value=0x2120 data=0xf2f2
value=0x0120 data=0x0000
value=0x2220 data=0x0e0e
value=0x0120 data=0x0000
value=0x2320 data=0x3535
value=0x0120 data=0x0000
value=0x2420 data=0x0606
value=0x0120 data=0x0000
value=0x2520 data=0x5050
value=0x0120 data=0x0000
value=0x2620 data=0x9c9c
value=0x0120 data=0x0000
value=0x2720 data=0x0d0d
value=0x0120 data=0x0000
value=0x2820 data=0x7171
value=0x0120 data=0x0000
value=0x2920 data=0x1111
value=0x0120 data=0x0000
value=0x2a20 data=0x1414
value=0x0120 data=0x0000
value=0x2b20 data=0x7171
value=0x0120 data=0x0000
value=0x2c20 data=0x7474
value=0x0120 data=0x0000
value=0x2d20 data=0x1919
value=0x0120 data=0x0000
value=0x2e20 data=0x4141
value=0x0120 data=0x0000
value=0x2f20 data=0xa5a5
value=0x0120 data=0x0000
value=0x1920 data=0x0505
value=0x0120 data=0x0000
value=0x9320 data=0xf0f0
value=0x0120 data=0x0000
value=0x9420 data=0x0f0f
value=0x0120 data=0x0000
value=0x1120 data=0x0000
value=0x0120 data=0x0000
value=0x0420 data=0x0000
value=0x0120 data=0x0000
value=0x6120 data=0x6060
value=0x0120 data=0x0000
value=0x0620 data=0x8080
value=0x0120 data=0x0000
value=0xb120 data=0x1b1b
value=0x0120 data=0x0000
value=0x0d20 data=0x8383
value=0x0120 data=0x0000
finished baseband init
rtlsdr_set_sample_rate
value=0x9f20 data=0x0e00
value=0x0120 data=0x0000
value=0xa120 data=0x0000
value=0x0120 data=0x0000
value=0x3f20 data=0x0000
value=0x0120 data=0x0000
value=0x3e20 data=0x0000
value=0x0120 data=0x0000
value=0x0120 data=0x1414
value=0x0120 data=0x0000
value=0x0120 data=0x1010
value=0x0120 data=0x0000
Sampling at 240000 S/s.
Device init done
Starting data capture
value=0x2148 data=0x1002
value=0x2148 data=0x0000
48.58 kBps
469.25 kBps
468.23 kBps
```

Hoel, 2020-09-01:
[image]
[image]

Hoel, 2020-09-01:
after a lot of struggle, IT WORKS!!! 

```
demod write read call 0x4c09
value=0x0120 data=0x10
value=0x0120 data=0x18
demod write read call 0x4c09
Samplerate OK 0
Sampling at 240000 S/s.
Device init done
Starting data capture
value=0x2148 data=0x1002
value=0x2148 data=0x0000
Bulk-in transfer from endpoint : 0x81
Bulk transfer 1
Bulk-in transfer from endpoint : 0x81
Bulk transfer 2
Bulk-in transfer from endpoint : 0x81
Bulk transfer 3
Bulk-in transfer from endpoint : 0x81
Bulk transfer 4
Bulk-in transfer from endpoint : 0x81
Bulk transfer 5
Bulk-in transfer from endpoint : 0x81
Bulk transfer 6
Bulk-in transfer from endpoint : 0x81
Bulk transfer 7
Bulk-in transfer from endpoint : 0x81
Bulk transfer 8
Bulk-in transfer from endpoint : 0x81
Bulk transfer 9
```

Hoel, 2020-09-01:
now i wonder what is the best approach to manage the bulk transfer buffer : acquisition is done in a 512 byte buffer (according to datasheet i can increase to 1023 but it still have to be proved) and i have to process the datas in realtime while continuing acquisition, which mean double / ping-pong buffering is required, my concern is that if i assign a different buffer at each call of lbk_bulk_read maybe there is a risk that some samples are lost? IS there another way?

ychuang3, 2020-09-02:
Hi,

The completion of a bulk transfer is depended on:
1. Transfer error occurred
2. Short or zero packet received. If the max. packet size is 512, any bulk-in packet < 512 bytes.
3. Transfer buffer full. 

You can give 1024 bytes bulk buffer or up to 16 KB. Host controller can collect bulk-in packets into your buffer.  

For example, you can give utr->data_len = 1024 and max. pktsz is 512, the bulk-in transfer will complete with the following bulk-in packet sequence (some examples)
 - 64 bytes
 - 510 bytes
 - 512 bytes + 18 bytes
 - error
 - 512 bytes + 0 bytes
 - 512 bytes + 512 bytes
  ....

To achieve or approximate ping-pong, it will be like the implementation of lbk interrupt-in transfer. Please refer to int_in_done() of lbk_driver.c. It issues a new transfer in the interrupt-in callback. (Note that all USB callback are called in interrupt context.)

Do the followings:
1. In lbk_bulk_read, use a static utr, instead of alloc_utr(), free_utr()
2. Implement a bulk_in_done() function (copy from int_in_done) 
3. In lbk_bulk_read, utr->func = bulk_in_done;
4.  Remove "utr->bIsTransferDone" pooling loop 
5. In bulk_in_done(), before invoking usbh_bulk_xfer() modify the buffer pointer "utr->buff" to another buffer (your ping-pong)

Y.C. Huang

Hoel, 2020-09-02:
Ah that is very interesting, i did not realize i could use bulk transfer with larger buffer than packet size, now i need to make some measurements to see how much time it will take to fill the 16K buffer, if its fast enough then it can be worth it

Hoel, 2020-09-02:
in an effort to optimize the BOM of this device, which is sadly already 108 parts, i want to share the crystal between baseband, RF frontend and M484, i plan to use a 16MHz Kyocera TCXO, is there any problem with USB to use a 16MHz cloc ksource instead of the default passive 12M source and is there any specific configuration (for STM32F4 there is), also is it wise to use the M484 clk-out buffer to feed the two other chips or should I directly share the crystal output between the 3 consumers? taking into account that phase noise is a concern in this application.

ychuang3, 2020-09-03:
Hi,

I think 16 MHz is OK, but we don't have such test ever.
Just modify  system_M480.h  
"#define __HXT       (12000000UL)    /*!< External Crystal Clock Frequency     */
as 16000000UL
 
USB required 48 MHz input clock, which comes from PLL (192MHz) / 4.

Y.C. Huang

ychuang3, 2020-09-03:
Hi,

USB 2.0 PHY requires 12 MHz clock. So, 12 MHz is MUST if you want to have USB 2.0 host.

Y.C. Huang

yachen, 2020-09-03:
Hi @Hoel ,

Neither "use the M484 clk-out buffer to feed the two other chips" nor "directly share the crystal output between the 3 consumers" is recommended by our H/W FAE.

Sincerely,

Yi-An Chen

Hoel, 2020-09-03:
HI,
Thank you for the answers, that is what i feared.

closed 2021-12-28 by ychuang3