OpenNuvoton/M480BSP #19: Do qspi and spim switch their io2 and io3 pins?

gavinfree, 2020-12-02:
Hi, I am checking qspi in 4-wire mode and spim using m484sidae and find their io2 and io3 pins may be switched. Is that right?
[image]
[image]

yachen, 2020-12-02:
Not sure what do yo mean by "switched". But the pin definition shown on the figures are correct.

gavinfree, 2020-12-02:
Well, I mean in QSPI, pin 26 mosi1 according to reference manual should be io2 in 4-wire spi, while in SPIM mode it is io3. That means they "switched".

yachen, 2020-12-02:
Then yes, they are switched.

gavinfree, 2020-12-02:
OK, thanks! By the way, the pin name of the qspi module is really confusing.

closed 2020-12-02 by gavinfree