OpenNuvoton/M480BSP #21: 483 CAN 发送后 收到一个数据全部为0的帧

wall-xv, 2021-04-28:
使用两个483的板子用CAN通信，发送后，总是收到一个数据全部为的包
UCAN0 TX ID= 400, Ft=1, Type=STD, DLC=5,Data=00,40,00,00,00,
UCAN0:0 RX ID= 0, Ft=0, Type=STD, DLC=0,Data= 这个帧数据全部为0
UCAN0:1 RX ID= 47E, Ft=0, Type=STD, DLC=0,Data=
这个问题有没有解决办法。

yachen, 2021-04-28:
測試時有用 CAN的 transceiver 嗎?

wall-xv, 2021-04-30:
现在使用两种transceiver
SN65HVD232DR 和SIT3051TK
CAN口插上线，就收到这个全0包
CAN口不插线，不会收到 全零包

mjchen1, 2021-05-03:
可以先確認一下,下面這些register,確定是否真的有收到message,
CAN_IIDR,CAN_NDAT1, CAN_NDAT2, CAN_IPND1, CAN_IPND1,
在收到中斷時,先打印出來看

wall-xv, 2021-05-14:
这个全0的message是只有发送端能收到. 接收端收不到.
这个全0的message,到时可以过滤掉.
有一个新的问题,483的CAN口怎样可以接收 Remote Frame, 远程帧. 按照手册内容
5. Dir = ‘1’ (direction = transmit), RmtEn = ‘0’ and UMask = ’1’
6. At the reception of a matching Remote Frame, the TxRqst bit of this Message Object is reset. 
The arbitration and control field (Identifier + IDE + RTR + DLC) from the shift register is stored 
in the Message Object of the Message RAM and the NewDat bit (CAN_IFn_MCON[15]) of this 
Message Object is set. The data field of the Message Object remains unchanged; the Remote 
Frame is treated similar to a received Data Frame.
设置, 也收不到Remote Frame, CANOpen协议需要使用Remote Frame, 请提供可以的接口.

wall-xv, 2021-05-14:
谢谢!
刚才解决了问题,
CAN_SetTxMsg中
        //tCAN->IF[u32MsgIfNum].MCON   =  CAN_IF_MCON_NEWDAT_Msk | pCanMsg->DLC | CAN_IF_MCON_TXIE_Msk | CAN_IF_MCON_EOB_Msk;
        改为tCAN->IF[u32MsgIfNum].MCON   =  CAN_IF_MCON_NEWDAT_Msk | pCanMsg->DLC  | CAN_IF_MCON_EOB_Msk;
中断就不会反馈发送message的Object 有新数据,这样就不会收到全0帧.
按手册修改CAN_SetTxMsg后,使用CAN_SetTxMsg设一个message Object,这样就可以收到Remote Frame了.

closed 2021-05-14 by wall-xv