OpenNuvoton/M480BSP #27: Cant run evaluationboard Nuvoton NuMaker-M483KG V1.1

TolunayOruc, 2022-04-06:
When running the sample code I get the following error in NuEclipse:

Open On-Chip Debugger 0.10.0-dev-00476-gd1de37d6-dirty (2021-12-08-10:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Error: We cannot find any Nuvoton device. Please check the hardware connection.
Error: If the ICE is used by another Nuvoton tool, please close the connection between the ICE and Nuvoton tool.
in procedure 'init' 
in procedure 'ocd_bouncer'

I have connected the evaluation board with usb and have set Nu-Link2-Me VCOM Switch No. 1 and 2 to 
ON. Building works fine but when I debug I get this error.

ccli0, 2022-04-07:
Hi TolunayOruc,

Please make sure the following items:
1. OpenOCD setup option is checked in the debugger tab of debug configurations and close other Nuvoton tool.
2. Please check the hardware connection.

If you still have problems, please fill in the -d3 in config options in the debugger tab of debug configurations to get more message.
-f ../scripts/interface/nulink.cfg -f ../scripts/target/numicroM4.cfg -d3

TolunayOruc, 2022-04-07:
> ** CAUTION: External email, be careful with links and attachments. **

Hi,

Following the user manual and using FirstExample and the delivered BSP480 template sample code I get the following error code with both projects:

Open On-Chip Debugger 0.10.0-dev-00476-gd1de37d6-dirty (2021-12-08-10:03)
Licensed under GNU GPL v2
For bug reports, read
      http://openocd.org/doc/doxygen/bugs.html
User : 13 1 command.c:544 command_print(): debug_level: 3
Debug: 14 2 options.c:166 add_default_dirs(): bindir=bin
Debug: 15 2 options.c:167 add_default_dirs(): pkgdatadir=
Debug: 16 2 options.c:168 add_default_dirs(): run_prefix=C:/Program Files (x86)/Nuvoton Tools/NuEclipse/V1.01.019/NuEclipse/eclipse/OpenOCD
Debug: 17 2 configuration.c:42 add_script_search_dir(): adding C:\Users\toruc\AppData\Roaming/OpenOCD
Debug: 18 3 configuration.c:42 add_script_search_dir(): adding C:/Program Files (x86)/Nuvoton Tools/NuEclipse/V1.01.019/NuEclipse/eclipse/OpenOCD/site
Debug: 19 3 configuration.c:42 add_script_search_dir(): adding C:/Program Files (x86)/Nuvoton Tools/NuEclipse/V1.01.019/NuEclipse/eclipse/OpenOCD/scripts
Debug: 20 5 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_gdb_port 3333
Debug: 21 5 command.c:143 script_debug(): command - gdb_port ocd_gdb_port 3333
Debug: 23 5 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_telnet_port 4444
Debug: 24 5 command.c:143 script_debug(): command - telnet_port ocd_telnet_port 4444
Debug: 26 6 configuration.c:82 find_file(): found C:/Program Files (x86)/Nuvoton Tools/NuEclipse/V1.01.019/NuEclipse/eclipse/OpenOCD/site/../scripts/interface/nulink.cfg
Debug: 27 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface hla
Debug: 28 8 command.c:143 script_debug(): command - interface ocd_interface hla
Debug: 30 8 command.c:364 register_command_handler(): registering 'ocd_hla_device_desc'...
Debug: 31 8 command.c:364 register_command_handler(): registering 'ocd_hla_serial'...
Debug: 32 8 command.c:364 register_command_handler(): registering 'ocd_hla_layout'...
Debug: 33 8 command.c:364 register_command_handler(): registering 'ocd_hla_close'...
Debug: 34 9 command.c:364 register_command_handler(): registering 'ocd_hla_reset'...
Debug: 35 9 command.c:364 register_command_handler(): registering 'ocd_hla_vid_pid'...
Debug: 36 9 command.c:364 register_command_handler(): registering 'ocd_hla_vids_pids'...
Debug: 37 9 command.c:364 register_command_handler(): registering 'ocd_hla_command'...
Debug: 38 9 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_layout nulink
Debug: 39 9 command.c:143 script_debug(): command - hla_layout ocd_hla_layout nulink
Debug: 41 10 hla_interface.c:239 hl_interface_handle_layout_command(): hl_interface_handle_layout_command
Debug: 42 10 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 43 10 command.c:143 script_debug(): command - hla_vids_pids ocd_hla_vids_pids 0x0416 0x511b 0x0416 0x511c 0x0416 0x511d
Debug: 45 10 hla_interface.c:312 hl_interface_handle_vids_pids_command(): hl_interface_handle_vids_pids_command
Debug: 46 10 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_hla_device_desc Nu-Link
Debug: 47 11 command.c:143 script_debug(): command - hla_device_desc ocd_hla_device_desc Nu-Link
Debug: 49 11 hla_interface.c:213 hl_interface_handle_device_desc_command(): hl_interface_handle_device_desc_command
Debug: 50 11 configuration.c:82 find_file(): found C:/Program Files (x86)/Nuvoton Tools/NuEclipse/V1.01.019/NuEclipse/eclipse/OpenOCD/site/../scripts/target/numicroM4.cfg
Debug: 51 12 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 52 12 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Info : 53 12 transport.c:285 jim_transport_select(): auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Debug: 54 12 hla_transport.c:191 hl_transport_select(): hl_transport_select
Debug: 55 12 command.c:364 register_command_handler(): registering 'ocd_hla'...
Debug: 56 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 57 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 58 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 59 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 60 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 61 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 62 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 63 13 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 64 14 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 65 14 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_delay'...
Debug: 66 14 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 67 14 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 68 14 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 69 14 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Debug: 70 15 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
[... cut ...]
Debug: 294 167 nulink_usb.c:1431 nulink_usb_close(): nulink_usb_close

I have used the user manual also for the setup of the debugger and this is mine setup:

***@***.***

With kind regards.

Tolunay Oruc
Mastervolt

This email message may contain confidential or privileged information. If you are not the intended recipient, please delete the message and any attachments and notify the sender by return email. You should not retain, distribute, disclose or use any of the information in this message.
**************************

ccli0, 2022-04-07:
Hi TolunayOruc,

Nu-Link2-Me cannot find any Nuvoton device on your PC. Do you check the hardware connection?
Or you can use KEIL or other Nuvoton tool to check the connection.
e.g. NuMicro ICP Programming Tool
https://www.nuvoton.com/tool-and-software/software-tool/programmer-tool/index.html

TolunayOruc, 2022-04-07:
Hi,

When using NuMicro ICP Programming Tool it crashes when I click on connect. On Keil it opens on default with M487 when opening the template project. This doesn’t match mine M483. When I change this it crashes when downloading to flash memory. On mine computer I can see the following:

***@***.***

https://www.nuvoton.com/tool-and-software/software-tool/programmer-tool/index.html<https://urldefense.proofpoint.com/v2/url?u=https-3A__www.nuvoton.com_tool-2Dand-2Dsoftware_software-2Dtool_programmer-2Dtool_index.html&d=DwMCaQ&c=tjPg1sbLxZ1bjgTYa5hhdyzvrRQ5lIm-1PubXARxg7w&r=V9-s_ACNT5doM3rSCG4bXx78kGqOOx-jU-w2VT6yMcw&m=crOjciPHDV3UvPNNEQnRMicyVdE_JMfUW9xtnwqQJ-6eAqV-X-ltOs4GCyjnm-_t&s=JmcJ-hRsRmKOqNEot1YUZqctVnUmmJ261vM7VqslApM&e=>

**************************

TolunayOruc, 2022-04-07:
Hi,

The problem was a broken usb cable. Because of that it was able to show up in device manager but didn’t connect in the tools. Thankyou for the fast reply and help!

With kind regards,
Tolunay Oruc
Mastervolt

https://www.nuvoton.com/tool-and-software/software-tool/programmer-tool/index.html<https://urldefense.proofpoint.com/v2/url?u=https-3A__www.nuvoton.com_tool-2Dand-2Dsoftware_software-2Dtool_programmer-2Dtool_index.html&d=DwMCaQ&c=tjPg1sbLxZ1bjgTYa5hhdyzvrRQ5lIm-1PubXARxg7w&r=V9-s_ACNT5doM3rSCG4bXx78kGqOOx-jU-w2VT6yMcw&m=crOjciPHDV3UvPNNEQnRMicyVdE_JMfUW9xtnwqQJ-6eAqV-X-ltOs4GCyjnm-_t&s=JmcJ-hRsRmKOqNEot1YUZqctVnUmmJ261vM7VqslApM&e=>

**************************

closed 2022-04-07 by TolunayOruc