OpenNuvoton/M480BSP #29: use FMC_WriteMultiple() will overwrite 8bytes

KunYi, 2022-06-06:
TRM says 
"The M480 series supports multi-word programming function to speed up Flash updated procedure.
The maximum programming length is up to 512 bytes, and the minimum programming length is 8
bytes (2 words). The multi-word programming is the fastest programming function if the programming
words more than 8 bytes, because only one set of Flash setup time and hold time needed for one time
operation. "

use Template Project of Keil MDK v5 to add the below code for verify FMC_WriteMultiple()

`

/* for verify FMC_WriteMultiple() */
__align(8) uint8_t buffer[512];
__align(8) uint8_t valid[512];

int main()
{
    uint32_t addr;
    uint32_t i;
    int32_t writeLen;
    uint32_t failed = 0;
    SYS_Init();

    /* Init UART to 115200-8n1 for print message */
    UART_Open(UART0, 115200);

    /* Connect UART to PC, and open a terminal tool to receive following message */
    printf("Verify Multi-word Program\n");
    SYS_UnlockReg();                   /* Unlock protected registers */

    FMC_Open();                        /* Enable FMC ISP function */

    FMC_ENABLE_AP_UPDATE();            /* Enable APROM erase/program */

    addr = FMC_APROM_END - FMC_FLASH_PAGE_SIZE; /* last page */
    FMC_Erase(addr);
    memset(buffer, 0x00, sizeof(buffer));
    for (i = 0; i < 256; i++)
    {
      buffer[i] = i;
    }

    writeLen = FMC_WriteMultiple(addr, (uint32_t *)buffer, 256); /* write 256bytes into FLASH */
    printf("write %d bytes\n", writeLen);
    memcpy(valid, (void *)addr, sizeof(buffer));
    for (i = 0; i < 256; i++)
    {
      if (buffer[i] != valid[i])
      {
        printf("write data incorrect at %d\n", i);
        failed = 1;
      }
    }

    if (valid[256] != 0xFF)  /* should keep in 0xFF */
    {
      printf("overwrite\n");
      failed = 1;
    }

    if (failed == 0)
      printf("testing passed\n");
    /* Got no where to go, just loop forever */
    while(1);

}
`
run the code on NuMaker-PFM-M487 and observe the valid[256] ~ valid[263] not keep 0xFF
[image]

KunYi, 2022-06-06:
verify code snippet on https://gist.github.com/KunYi/25d68dca1cc430ff1cce2041c7b61cae

KunYi, 2022-06-10:
Hi Jacky,
     thank you for your reply, but I test with  bsp\SampleCode\StdDriver\FMC_MultiWordProgram still failed
     the MultiWord Program still write 256+8bytes when call "FMC_WriteMultiple(addr, (uint32_t *)buffer, 256);" in SRAM
     I modify "**multi_word_prog.c**" for testing and re-build to generation 
     "**multi_word_prog.bin**" then to build main_loader project and flash on NuMaker-PFM-M487, 
     still got same result
    my multi_word_prog.c 
```

/**************************************************************************//**
 * @file     multi_word_prog.c
 * @version  V1.00
 * @brief    This sample run on SRAM to show FMC multi word program function.
 *
 *
 * @copyright (C) 2016 Nuvoton Technology Corp. All rights reserved.
*****************************************************************************/
#include <stdio.h>
#include <string.h>
#include "NuMicro.h"

#define USE_DRIVER_API

uint32_t    page_buff[FMC_FLASH_PAGE_SIZE/4];

void SYS_Init(void)
{
    /* Unlock protected registers */
    SYS_UnlockReg();

    /* Set XT1_OUT(PF.2) and XT1_IN(PF.3) to input mode */
    PF->MODE &= ~(GPIO_MODE_MODE2_Msk | GPIO_MODE_MODE3_Msk);

    /* Enable HXT clock */
    CLK_EnableXtalRC(CLK_PWRCTL_HXTEN_Msk);

    /* Wait for HXT clock ready */
    CLK_WaitClockReady(CLK_STATUS_HXTSTB_Msk);

    /* Switch HCLK clock source to HXT */
    CLK_SetHCLK(CLK_CLKSEL0_HCLKSEL_HXT,CLK_CLKDIV0_HCLK(1));

    /* Set core clock as PLL_CLOCK from PLL */
    CLK_SetCoreClock(FREQ_192MHZ);

    /* Set both PCLK0 and PCLK1 as HCLK/2 */
    CLK->PCLKDIV = CLK_PCLKDIV_APB0DIV_DIV2 | CLK_PCLKDIV_APB1DIV_DIV2;

    /* Enable UART module clock */
    CLK_EnableModuleClock(UART0_MODULE);

    /* Select UART module clock source as HXT and UART module clock divider as 1 */
    CLK_SetModuleClock(UART0_MODULE, CLK_CLKSEL1_UART0SEL_HXT, CLK_CLKDIV0_UART0(1));

    /* Update System Core Clock */
    SystemCoreClockUpdate();

    /* Set GPB multi-function pins for UART0 RXD and TXD */
    SYS->GPB_MFPH &= ~(SYS_GPB_MFPH_PB12MFP_Msk | SYS_GPB_MFPH_PB13MFP_Msk);
    SYS->GPB_MFPH |= (SYS_GPB_MFPH_PB12MFP_UART0_RXD | SYS_GPB_MFPH_PB13MFP_UART0_TXD);

    /* Lock protected registers */
    SYS_LockReg();
}

void UART0_Init(void)
{
    /* Configure UART0 and set UART0 baud rate */
    UART_Open(UART0, 115200);
}

__align(8) uint8_t buffer[512];
__align(8) uint8_t valid[512];

void testMultiWordProgram(void) {
    int32_t writeLen;
    uint32_t addr;
  uint32_t failed = 0;
  uint32_t i;

  SYS_UnlockReg();                   /* Unlock protected registers */

  FMC_Open();                        /* Enable FMC ISP function */

  FMC_ENABLE_AP_UPDATE();            /* Enable APROM erase/program */

    /* prepare test data */
    memset(buffer, 0x00, sizeof(buffer));
  for (i = 0; i < 256; i++)
        buffer[i] = i;

  addr = FMC_APROM_END - FMC_FLASH_PAGE_SIZE; /* last page */
  FMC_Erase(addr);

  writeLen = FMC_WriteMultiple(addr, (uint32_t *)buffer, 256);

  printf("write %d bytes to addr:0x%p\n", writeLen, (void*)addr);
  memcpy(valid, (void *)addr, sizeof(buffer));
  for (i = 0; i < 256; i++)
  {
      if (buffer[i] != valid[i])
      {
        //printf("write data incorrect at %d\n", i);
        failed = 1;
      }
  }

  if (valid[256] != 0xFF)
  {
        printf("overwrite:");
        for (i = 0; i < 16; i++)
            printf(" %02x", valid[256+i]);

        printf("\n");
    failed = 1;
  }

  if (failed == 0)
     printf("testing passed\n");
  /* Got no where to go, just loop forever */
  while(1);
}

int main()
{
    uint32_t  i, addr, maddr;          /* temporary variables */

    SYS_Init();                        /* Init System, IP clock and multi-function I/O */

    UART0_Init();                      /* Initialize UART0 */

    printf("\n\n");
    printf("+-------------------------------------+\n");
    printf("|    M480 Multi-word Program Sample   |\n");
    printf("+-------------------------------------+\n");
        testMultiWordProgram();
    while (1);
}

```

and result on the below

```
+-------------------------------------+
|    M480 Multi-word Program Sample   |
+-------------------------------------+
write 256 bytes to addr:0x0007f000
overwrite: 00 00 00 00 00 00 00 00 ff ff ff ff ff ff ff ff

```

KunYi, 2022-06-10:
or just change programming length to 256 from 512, like the below picture
in  bsp\SampleCode\StdDriver\FMC_MultiWordProgram

[image]

will got the below result
[image]

ychuang3, 2022-06-27:
Hi,
It's a bug of FMC_WriteMultiple() API. 
Please update the latest patch to fix it.

Thank you,
Jacky

KunYi, 2022-06-27:
@ychuang3  
Thanks! Jacky.
verify the API work correctly now!