OpenNuvoton/M480BSP #33: Question about the use of extern in Library/StdDriver/inc/fmc.h header file

cy023, 2023-03-02:
Hi there,

I have a question about the use of `extern` in the Library/StdDriver/inc/fmc.h header file. Specifically, I noticed that the `extern` keyword is used in function declarations. However, as far as I know, C already assumes extern by default when declaring functions.

I would like to understand why `extern` is used in these function declarations, and whether it is necessary.

Thanks in advance for any clarification you can provide!

sadikthanos, 2023-12-15:
Hi @cy023  your observation is correct.

In C, function declarations are considered implicitly extern by default. When you declare a function in C, it is assumed to have external linkage, meaning that it can be accessed by other source files.

Rather I would believe that this might be an Coding style choice or just to maintain consistency in all of the declarations in the header files.

As  it doesn't alter any functionality of that feels like just bathing in the rain :)

ychuang3, 2023-12-16:
A very interesting and apt metaphor.