OpenNuvoton/M480BSP #35: USB Host connect->disconnect->connect on some devices

jeromeDms, 2024-02-14:
Hi
I'm observing a weird issue on a USB device (MIDI) while using your USB Host library.
I checked with and without a hub and the result is the same.
The lib has a weird bahaviour when some device are connected, I observe the connection, then a disconnection then a new connection and the device finally works.
When trying to debug the above issue, I found the ohci requests a disconnection, this happens in ohci_rh_polling

```
/*--------------------------------------------------------------------------------*/
/*  connect status change                                                         */
/*--------------------------------------------------------------------------------*/

_ohci->HcRhPortStatus[i] = USBH_HcRhPortStatus_CSC_Msk;     /* clear CSC          */

 if (_ohci->HcRhPortStatus[i] & USBH_HcRhPortStatus_CCS_Msk)
 {
            /*----------------------------------------------------------------------------*/
            /*  First of all, check if there's any previously connected device.           */
            /*----------------------------------------------------------------------------*/
           while (1)
            {
                udev = ohci_find_device_by_port(i+1);
                if (udev == NULL)
                    break;
                disconnect_device(udev);  <<<<<< DISCONNECT HERE
            }
```

Any idea why this could happen ?
Also, while looking at the above source code, there is something I cannot understand :

_ohci->HcRhPortStatus[i] = USBH_HcRhPortStatus_CSC_Msk; 
Followed by :
 if (_ohci->HcRhPortStatus[i] & USBH_HcRhPortStatus_CCS_Msk)
 
 How the if() above could be true since the previous line assigns USBH_HcRhPortStatus_CSC_Msk to _ohci->HcRhPortStatus[I] ?
 If the idea is to clear the CSC bit only,  should the first line above be :
 
  _ohci->HcRhPortStatus[i] &= ~USBH_HcRhPortStatus_CSC_Msk; 

Thanks for any help you can provide.

ychuang3, 2024-02-15:
USBH_HcRhPortStatus_CSC_Msk   is  "Connect Status Changed", but
USBH_HcRhPortStatus_CCS_Msk  is "Current Connect Status".

jeromeDms, 2024-02-15:
Thanks for your answer.

Regarding the fist part of my post, can you give some tips why some of the connected devices are connecting then disconnecting and reconnecting again ?
What could cause this behavior on some devices ?

ychuang3, 2024-02-15:
It could be a poor cable connection or the device itself might be disconnecting and reconnecting. You can see how the code is written - connection and disconnection handling only occur when there is a connect status change, determined by the Host Controller H/W port status.