OpenNuvoton/M480BSP #36: ISP_DFU_20 driver transfer file in case of USB 2.0 (NUVOTON M487KM) driver development

SANKETMODHE7028, 2024-02-19:
1 . i have to transmit or receiving the data . so how to do this?
2. i have to change in DFU_Transfer file ?
3. void HSUSBD_ProcessSetupPacket(void)
{
    /* Setup packet process */
    gUsbCmd.bmRequestType = (uint8_t)(HSUSBD->SETUP1_0 & 0xfful);
    gUsbCmd.bRequest = (uint8_t)((HSUSBD->SETUP1_0 >> 8) & 0xfful);
    gUsbCmd.wValue = (uint16_t)HSUSBD->SETUP3_2;
    gUsbCmd.wIndex = (uint16_t)HSUSBD->SETUP5_4;
    gUsbCmd.wLength = (uint16_t)HSUSBD->SETUP7_6;

    /* USB device request in setup packet: offset 0, D[6..5]: 0=Standard, 1=Class, 2=Vendor, 3=Reserved */
    switch(gUsbCmd.bmRequestType & 0x60ul)
    {
    case REQ_STANDARD:
    {

        HSUSBD_StandardRequest();
        break;
    }

    case REQ_CLASS:
    {
        DFU_ClassRequest();
        break;
    }

4.In hsusbd_user.c   file only  HSUSBD_StandardRequest()  function is working for configuration part . when  DFU_ClassRequest() ; function is called? 
5. suppose i have to pass data so I have to calling seperately?
6. please give me the proper logic.

ychuang3, 2024-02-21:
I'm not quite clear on your point of concern. You don't need to trace the driver unless you want to write a separate driver yourself. Using this sample, please refer to the documentation under the WindowsTool directory.

SANKETMODHE7028, 2024-02-21:
ok ........ 
only give me idea when the DFU_CLASS_REQUEST function is called?
and how to buffer read or write?

SANKETMODHE7028, 2024-02-21:
> I'm not quite clear on your point of concern. You don't need to trace the driver unless you want to write a separate driver yourself. Using this sample, please refer to the documentation under the WindowsTool directory.

ok ........
only give me idea when the DFU_CLASS_REQUEST function is called?
and how to buffer read or write?

ychuang3, 2024-02-21:
DFU_ClassRequest() is called when DFU class command received from USB Host (Windows PC).
Data is from USB Host and directly programmed to flash., refer to the WriteData().

Could you explain the purpose of wanting to trace the code? Thank you.

ychuang3, 2024-02-21:
The USB device is passive, and all data transmission is initiated by the USB Host. Therefore, you must perform read operations from the Windows PC side for the USB device to receive commands and carry out corresponding data transfers. We do not provide the source code for the Windows side; this sample is intended for firmware updates. What is the purpose of sending data to the PC side? There may be other suitable sample codes, but since you haven't clearly explained your purpose, I'm unable to provide specific helpful suggestions.

SANKETMODHE7028, 2024-02-21:
> The USB device is passive, and all data transmission is initiated by the USB Host. Therefore, you must perform read operations from the Windows PC side for the USB device to receive commands and [...]

I am not getting properly concept. do you have any  proper datasheet on DFU ?
and how to handle this please explain first..

actually  i am  working on project so my task is to transmit or receive the data . using DFU_transfer file .

ychuang3, 2024-02-21:
You can find DFU spec. by Google search...

Looks like your project is not on Nuvoton M480 platform, right?
I think you don't even run the sample code.

SANKETMODHE7028, 2024-02-21:
controller is NUVOTON M487KM .

actually totally working in background only   HSUSBD_StandardRequest(); this function run for configuration part. 

URB 165 URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE    Down  15:22:29.574 18 bytes IN PENDING DISPATCH_LEVEL  
URB 166 URB_FUNCTION_CONTROL_TRANSFER                       Up       15:22:29.585 18 bytes IN SUCCESS DISPATCH_LEVEL  
URB 167 URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE    Down  15:22:29.585  9 bytes   IN   PENDING    DISPATCH_LEVEL  
URB 168 URB_FUNCTION_CONTROL_TRANSFER                      Up       15:22:29. 597  9 bytes   IN   SUCCESS    DISPATCH_LEVEL  
URB 169 URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE   Down  15:22:29.597    27 bytes  IN   PENDING   DISPATCH_LEVEL  
URB 170 URB_FUNCTION_CONTROL_TRANSFER                     Up        15:22:29.609   27 bytes  IN   SUCCESS   DISPATCH_LEVEL  
URB 171 URB_FUNCTION_SELECT_CONFIGURATION            Down     15:22:29.609   27 bytes  OUT  DISPATCH_LEVEL  
URB 172 URB_FUNCTION_SELECT_CONFIGURATION           Up            15:22:29.621   27 bytes  IN    SUCCESS DISPATCH_LEVEL  

only this is  showing in usb analyser

SANKETMODHE7028, 2024-02-21:
> You can find DFU spec. by Google search...
> 
> Looks like your project is not on Nuvoton M480 platform, right? I think you don't even run the sample code.

i searched this but how to use in my project?

SANKETMODHE7028, 2024-02-21:
> > You can find DFU spec. by Google search...
> > Looks like your project is not on Nuvoton M480 platform, right? I think you don't even run the sample code.
> 
> i searched this but how to use in my project?

ychuang3, 2024-02-22:
The link to USB org including DFU spec.
https://www.usb.org/documents?search=&type%5B0%5D=55&items_per_page=50

Please find:
[image]

HSUSBD_StandardRequest() is for standard request, that is, for any USB device. DFU data transfer use class commands, not standard commands.

Do you run the ISP tool or any program supports DFU on the host PC?

SANKETMODHE7028, 2024-02-22:
> The link to USB org including DFU spec. https://www.usb.org/documents?search=&type%5B0%5D=55&items_per_page=50
>
> Please find: [...]

I have Numicro ICP programing tool utility .
and in my project EL series utility are used . for that host (pc) sending first command to slave (device)..
so how to use this class command?

SANKETMODHE7028, 2024-02-22:
> I have Numicro ICP programing tool utility . and in my project EL series utility are used . for that host (pc) sending first command to slave (device).. so how to use this class command?

i am using this DFU file and discriptor file . only changed in VID and PID for HMI detection . remaining code is same . so what i do first step?

ychuang3, 2024-02-22:
Not ICP tool, please use ISP tool which is for ISP_XXX samples.

SANKETMODHE7028, 2024-02-22:
> Not ICP tool, please use ISP tool which is for ISP_XXX samples.

Do you have link of ICP tool? please send

SANKETMODHE7028, 2024-02-23:
> > Not ICP tool, please use ISP tool which is for ISP_XXX samples.
> 
> Do you have link of ISP tool? please send

I downloaded ISP tool , suppose i start this what happened in next step ? directly calling DFU_class_request function ?

ychuang3, 2024-02-23:
Sorry, ISP is for USBD HID, not DFU.
For DFU, just run the tool in BSP  bsp\SampleCode\ISP\ISP_DFU_20\WindowsTool.
Please read the PDF in this folder.

SANKETMODHE7028, 2024-02-23:
> Sorry, ISP is for USBD HID, not DFU. For DFU, just run the tool in BSP bsp\SampleCode\ISP\ISP_DFU_20\WindowsTool. Please read the PDF in this folder.

hello sir , actually in my case my pc is host . first pc sending 1 byte . so how to response to this  ? how to know 1 byte receive in device?

ychuang3, 2024-02-23:
Do you run dfu-util.exe on the host PC? 
If not, what's the host application?

SANKETMODHE7028, 2024-02-23:
> Do you run dfu-util.exe on the host PC? If not, what's the host application?

Dear sir,
We have our custom USB communication Utility at PC side, with this PC utility we are communicating our sensor reading to host PC. Our PC  utility is implemented by basic byte communication and sending data over 64 byte packet format. Kindly help to check received byte and send byte to host PC utility.

SANKETMODHE7028, 2024-02-23:
> What's the USB device class supported by your PC utility? If it does not support DFU class device , you will never receive class commands. Does you PC utility found M487? You log messages are standard transfers done by Windows, not your PC utility.

Dear sir,
We have our own HMI drivers for our product communication. We have already implemented HMI protocol setup with different controller of NXP, they are working OK with HMI Based PC utility. 
Currently we are porting our Nuvoton controller for NXP, hence same HMI protocol is need to be implemented in Nuvoton M487 controller, so that same PC software will support Nuvoton communication also.
Currently we have selected Base code of nuvoton from BSP base as ISP, we have modifed VID and PID and configuration updated in ISP code, with this changes now Nuvoton detecting as HMI port in our PC, now next step is to communicate with PC host for this purpose we need to send and receive byte and packet, can you please explain how we can use ISP existing function to write and read byte from Host PC.
If same HMI protocol not possible with our shortlisted ISP base code then kindly suggest another BSP code which can fullfill our requirment.

ychuang3, 2024-02-23:
If you modify PID and VID of any M487 sample code, your PC utility can find it, too.
But, you cannot communicate with each other.
For example, if you modify the mass storage USB device PID and VID, your PC utility can find it, too. 
However, they cannot communicate with each other, because you PC utility does not run mass storage class.

So, I am asking what's the USB device class your PC utility is running. 
If it is not running DFU class, select ISP_DFU sample is a wrong choice.

If you cannot tell what's the USB device class of you PC utility, we cannot go further.

SANKETMODHE7028, 2024-02-23:
> If you modify PID and VID of any M487 sample code, your PC utility can find it, too. But, you cannot communicate with each other. For example, if you modify the mass storage USB device PID and VID, [...]

Dear sir,
In our last product we used NXP LPC controller and PC side HMI communication done with CDC class.  PC utility uses CDC class request to send and receive data from controller.

ychuang3, 2024-02-24:
OK, obviously ISP_DFU20 is not CDC.
You can try HSUSBD_VCOM_SerialEmulator sample code. VCOM is a subclass of CDC.
I am not sure your PC utility is running VCOM or not.
Can you connect the NXP device to Windows and then check "Device Manager" to see what type of device presented?

SANKETMODHE7028, 2024-02-26:
> OK, obviously ISP_DFU20 is not CDC. You can try HSUSBD_VCOM_SerialEmulator sample code. VCOM is a subclass of CDC. I am not sure your PC utility is running VCOM or not. Can you connect the NXP device to Windows and then check "Device Manager" to see what type of device presented?

>ur PC utility is running VCOM or not.

Dear sir,
Will check HSUSBD_VCOM sample code and update you.
Our PC utility using HMI COM port, we have purchased USB distribution for our company, when we connect our product to PC then we install our PC drivers, then in device manager they will represent as "HMI USB device" in USB port list and in driver detail are company name list listed.

For NXP devices different USB drivers are available like HID, CDC and VCOM, when they are connected to PC then they represented as base driver used for testing, we our previous version we shortlisted CDC driver example and modified its endpoints and descriptor as per our HMI driver required and implemented same communication with PC with USB port of NXP controller.

Same communication we need to implement in Nuvoton M487KMCAN controller also, as we have our own HMI drivers and PC utility are already developed, hence we just need to modify Nuvoton M487KMCAN drivers to match our existing HMI protocol.

ychuang3, 2024-02-26:
Since it requires installing an additional driver, it seems there must be some custom handshake protocol involved.
Unless you are familiar with the specifics of these protocols, the only method I can think of is to intercept the USB transfers between the NXP device and its PC utility to replicate its transmission scenario.

ychuang3, 2024-02-26:
Bu the USB analyzer, can you see the first 9 bytes send from the PC utility?

SANKETMODHE7028, 2024-02-26:
> Bu the USB analyzer, can you see the first 9 bytes send from the PC utility?

NO fist byte not seen

SANKETMODHE7028, 2024-02-26:
> > Bu the USB analyzer, can you see the first 9 bytes send from the PC utility?
> 
> NO first byte not seen

ychuang3, 2024-02-26:
So the PC utility isn't actually performing any real data transfer at all. It seems to have some checking conditions, evidently the M480 USB device settings don't meet the criteria.

SANKETMODHE7028, 2024-02-26:
> So the PC utility isn't actually performing any real data transfer at all. It seems to have some checking conditions, evidently the M480 USB device settings don't meet the criteria.

this is inturupt handler
 if (IrqSt & HSUSBD_CEPINTSTS_SETUPPKIF_Msk)//setup packet has been received from host
        {

            HSUSBD_CLR_CEP_INT_FLAG(HSUSBD_CEPINTSTS_SETUPPKIF_Msk);
            HSUSBD_ProcessSetupPacket(); /// setup packet.....

            return;
        }
can u please explain what happened in this case .

ychuang3, 2024-02-27:
First, you need to make sure that the PC utility has sent a request to the USB device to request the transmission of the first byte of data you mentioned. Have you done this? Please check your USB analyzer log.
It seems that you don't understand what I've been explaining. Your PC utility doesn't think this is a USB device that it supports, so it doesn't initiate any data transfer. Therefore, you won't get the first byte you expect on the USB device side no matter what. First, you need to make the PC utility think it's the correct USB device. My suggestion is to modify the M487 USB descriptors and completely clone the descriptors of the NXP device, including the device, configuration, interface, and endpoint descriptors.

SANKETMODHE7028, 2024-02-27:
> First, you need to make sure that the PC utility has sent a request to the USB device to request the transmission of the first byte of data you mentioned. Have you done this? Please check your USB [...]

ok i will check configuration part first . and also usb initialization part.

SANKETMODHE7028, 2024-02-27:
> ok i will check configuration part first . and also usb initialization part.

)

just usb connect to pc i got this at usb analyser .
000007: Control Transfer (UP), 2024-02-27 12:42:50.1708152 +0.0120191. (1. Device: Nu_DFU) Status: 0x00000000
Pipe Handle: Control Pipe
 09 02 1B 00 01 01 00 C0 50                        .......ÀP
Setup Packet
 80 06 00 02 00 00 09 00                           €.......
Recipient: Device
Request Type: Standard
Direction: Device->Host
Request: 0x6 (GET_DESCRIPTOR)
Value: 0x200
Index: 0x0
Length: 0x9

ychuang3, 2024-02-27:
This is a standard request. Whenever you connect any type of USB device (pen drive, wifi dongle, mouse, ...), you'll see this type of request. So, you haven't made any progress.

SANKETMODHE7028, 2024-02-28:
> device, configuration, interface, and endpoint descriptors.

i also checked and modified device, configuration, interface, and endpoint descriptors. not working.

SANKETMODHE7028, 2024-02-28:
> First, you need to make sure that the PC utility has sent a request to the USB device to request the transmission of the first byte of data you mentioned. Have you done this? Please check your USB [...]

i also checked and modified device, configuration, interface, and endpoint descriptors. not working.

SANKETMODHE7028, 2024-02-29:
> First, you need to make sure that the PC utility has sent a request to the USB device to request the transmission of the first byte of data you mentioned. Have you done this? Please check your USB [...]

any other option for this ? because i modify these . its not working .

ychuang3, 2024-02-29:
What device presented on your Windows PC Device Manager?

SANKETMODHE7028, 2024-02-29:
> What device presented on your Windows PC Device Manager?

HMI USB Device

ychuang3, 2024-02-29:
> > What device presented on your Windows PC Device Manager?
> 
> HMI USB Device

Are you saying that when the M487 USB device is connected to your PC, your device manager shows "HMI USB Device"? If so, you should start receiving some requests other than standard ones. So, please use a USB analyzer to inspect what command the PC utility issues after the Select Configuration request.

SANKETMODHE7028, 2024-02-29:
> Are you saying that when the M487 USB device is connected to your PC, your device manager shows "HMI USB Device"? If so, you should start receiving some requests other than standard ones. So, please [...]

after the standard request not any command received .i checked.

SANKETMODHE7028, 2024-03-01:
> Are you saying that when the M487 USB device is connected to your PC, your device manager shows "HMI USB Device"? If so, you should start receiving some requests other than standard ones. So, please [...]

Can i used VCOM UTILITY ?

ychuang3, 2024-03-02:
Why not give it a try? The M480 BSP has VCOM sample code. However, the HMI USB device is not a standard USB class device defined by USB officials. I believe it might have some custom commands or transmission behaviors, which could hinder communication with a VCOM device.

SANKETMODHE7028, 2024-03-05:
> Why not give it a try? The M480 BSP has VCOM sample code. However, the HMI USB device is not a standard USB class device defined by USB officials. I believe it might have some custom commands or transmission behaviors, which could hinder communication with a VCOM device.

I integrated VCOM utility but not working . please tell me how to cheacking which command is coming on usb analyser.. because its only showing standard request configuration .  cammand can we see on analyser? .

ychuang3, 2024-03-05:
Are you referring to the VCOM utility as a PC-side tool here? If so, did you use the VCOM sample on the M487?

SANKETMODHE7028, 2024-03-05:
> Are you referring to the VCOM utility as a PC-side tool here? If so, did you use the VCOM sample on the M487?

instead of ISP_DFU_20 we are used VCOM .  at PC side already driver installed(pc utility )developed by .net team members in my company. now in this case HMI not detecting i also changed USB setting also . 

my meeting also done with campas componant team . he told me DFU is ok  in this case.

ychuang3, 2024-03-05:
So your PC side is still using the HMI USB and hasn't switched to the VCOM utility?
You should be using the NXP HMI USB device. Try another approach: use a USB analyzer to fully capture the transfer content between it and the PC utility. Then, replicate the process exactly on the M487. This should at least get you to the data transfer stage.

SANKETMODHE7028, 2024-03-07:
> So your PC side is still using the HMI USB and hasn't switched to the VCOM utility? You should be using the NXP HMI USB device. Try another approach: use a USB analyzer to fully capture the transfer [...]

yes integrating with VCOM.  port not detecting . Unknown USB Device (Device Discriptor Request failed) showing in Device manager. 

I have to only modify to VCOM_TransferData(void); function right? not modify to  VCOM_ClassRequest(void) ;.

ychuang3, 2024-03-07:
"Unknown USB Device" indicates that there is an issue somewhere. It should be normal to connect your PC using the original VCOM sample from M480 BSP, and you should be able to see the COMx device in Device Manager.

SANKETMODHE7028, 2024-03-08:
> "Unknown USB Device" indicates that there is an issue somewhere. It should be normal to connect your PC using the original VCOM sample from M480 BSP, and you should be able to see the COMx device in Device Manager.

one doubt sir, suppose I modify the code ( class request function ) then it will effect on  whole code?

ychuang3, 2024-03-11:
You didn't answer my question. I can't make a judgment.

SANKETMODHE7028, 2024-03-16:
[image]
hello sir my IDE have some problem its not working . how to resolve please help me.

ccli0, 2024-03-18:
IDE cannot recognize your Nu-Link USB device.
Please confirm whether the connection between Nu-Link and M487 target chip is correct.
Please make sure the debugging settings are as follows
[image]

SANKETMODHE7028, 2024-03-18:
> IDE cannot recognize your Nu-Link USB device. Please confirm whether the connection between Nu-Link and M487 target chip is correct. Please make sure the debugging settings are as follows [...]

Same setting is there in my NU eclipse. i checked all setting . but problem not resolved.

SANKETMODHE7028, 2024-03-18:
> > IDE cannot recognize your Nu-Link USB device. Please confirm whether the connection between Nu-Link and M487 target chip is correct. Please make sure the debugging settings are as follows ![image](https://private-user-images.githubusercontent.com/47807020/313525359-5c436e84-fb57-41de-ae50-56f9bd705d47.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3MzQxMjgsIm5iZiI6MTcxMDczMzgyOCwicGF0aCI6Ii80NzgwNzAyMC8zMTM1MjUzNTktNWM0MzZlODQtZmI1Ny00MWRlLWFlNTAtNTZmOWJkNzA1ZDQ3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE4VDAzNTAyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU5MzQ1MmU2M2E5ZWM3Mjk5NmE4MzMwNjAwZWZjYzNlYzVhOTc3NWQ1MDIzMjM5NDc0MTU4OWYxMGZmOGFmNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Xd7abbC4zLMnvhhB85UXQFTWIHK5ie7vn7nas3R_GFA)
> 
> Same setting is there in my NU eclipse. i checked all setting . but problem not resolved. i also checked in NU micro ICP programming tool . in this case its connected fine.

[image]

ccli0, 2024-03-18:
Please download and use NuMicro ICP Programming Tool 3.15.7623r.exe to check if your Nu-Link connection is correct.
https://www.nuvoton.com/tool-and-software/software-tool/programmer-tool/

SANKETMODHE7028, 2024-03-18:
[image]

SANKETMODHE7028, 2024-03-18:
[image]

now using anather pc .

these is showing.

ccli0, 2024-03-18:
1. NU micro ICP programming tool version?
2. NuEclipse version?
3. Nu-Link_USB_Driver installed?

[51 more replies]