OpenNuvoton/M480BSP #42: Nueclipse gets stuck when flashing the image.

prometheans152, 2024-09-11:
I followed the NuEclipse User Manual to create a new project and successfully built the ELF file, but when I try to run it and flash the ELF into the M483KG, it gets stuck halfway through. The log is as follows:

**Open On-Chip Debugger 0.10.022.0-dev-00493-gd6db465b-dirty (2023-12-14-15:04)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : NULINK is Nu-Link2
Info : NULINK firmware_version(7691), product_id(0x00d4851f)
Info : IDCODE: 0x2BA01477
Info : NuMicro.cpu: hardware has 6 breakpoints, 4 watchpoints
Started by GNU ARM Eclipse
Info : accepting 'gdb' connection on tcp/3333
Info : Device ID: 0x01348344
Info : Device Name: M483KGCAE2A
Info : bank base = 0x00000000, size = 0x00040000
Info : Device ID: 0x01348344
Info : Device Name: M483KGCAE2A
Info : bank base = 0x0001f000, size = 0x00000000
Info : Device ID: 0x01348344
Info : Device Name: M483KGCAE2A
Info : bank base = 0x00100000, size = 0x00001000
Info : Device ID: 0x01348344
Info : Device Name: M483KGCAE2A
Info : bank base = 0x00300000, size = 0x00000010
Info : Device ID: 0x01348344
Info : Device Name: M483KGCAE2A
Warn : NuMicro flash driver: Failed to detect flash size
NuMicro.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000009b4 msp: 0x20020000
numicro chip_erase complete
NuMicro.cpu: target state: halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0xffffffd8
Info : numicro write_isp 0x00300008 0xffff5a5a
Info : Nuvoton NuMicro: Flash Write ...
Info : Program at 0x00000000 to 0x00001e33
wrote 7732 bytes from file C:\Users\user\eclipse-workspace\EX\Debug\EX.hex in 0.786921s (9.595 KiB/s)
NuMicro.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000009b4 msp: 0x20020000
===== arm registers
(0) r0 (/32): 0x00000000
(1) r1 (/32): 0x00000000
(2) r2 (/32): 0x00000000
(3) r3 (/32): 0x00000000
(4) r4 (/32): 0x00000000
(5) r5 (/32): 0x00000000
(6) r6 (/32): 0x00000000
(7) r7 (/32): 0x00000000
(8) r8 (/32): 0x00000000
(9) r9 (/32): 0x00000000
(10) r10 (/32): 0x00000000
(11) r11 (/32): 0x00000000
(12) r12 (/32): 0x00000000
(13) sp (/32): 0x20020000
(14) lr (/32): 0xFFFFFFFF
(15) pc (/32): 0x000009B4
(16) xPSR (/32): 0x01000000
(17) msp (/32): 0x20020000
(18) psp (/32): 0x00000000
(19) primask (/1): 0x00
(20) basepri (/8): 0x00
(21) faultmask (/1): 0x00
(22) control (/2): 0x00
(23) d0 (/64): 0x0000000000000000
(24) d1 (/64): 0x0000000000000000
(25) d2 (/64): 0x0000000000000000
(26) d3 (/64): 0x0000000000000000
(27) d4 (/64): 0x0000000000000000
(28) d5 (/64): 0x0000000000000000
(29) d6 (/64): 0x0000000000000000
(30) d7 (/64): 0x0000000000000000
(31) d8 (/64): 0x0000000000000000
(32) d9 (/64): 0x0000000000000000
(33) d10 (/64): 0x0000000000000000
(34) d11 (/64): 0x0000000000000000
(35) d12 (/64): 0x0000000000000000
(36) d13 (/64): 0x0000000000000000
(37) d14 (/64): 0x0000000000000000
(38) d15 (/64): 0x0000000000000000
(39) fpscr (/32): 0x00000000
===== Cortex-M DWT registers
(40) dwt_ctrl (/32)
(41) dwt_cyccnt (/32)
(42) dwt_0_comp (/32)
(43) dwt_0_mask (/4)
(44) dwt_0_function (/32)
(45) dwt_1_comp (/32)
(46) dwt_1_mask (/4)
(47) dwt_1_function (/32)
(48) dwt_2_comp (/32)
(49) dwt_2_mask (/4)
(50) dwt_2_function (/32)
(51) dwt_3_comp (/32)
(52) dwt_3_mask (/4)
(53) dwt_3_function (/32)**

ccli0, 2024-09-11:
Did you see the following screen when you enter debugging?
[image]
What is the version of NuEclipse?
Are you using Windows or Ubuntu? What is the version?

prometheans152, 2024-09-11:
Sorry, I made a mistake. The project created using the NuEclipse User Manual runs successfully, but the sample code from the BSP doesn't work, as shown in the image.
[image]
By the way, which programming software should be used to flash the built ELF file? It seems like NuMicro ICP can't flash ELF files.

prometheans152, 2024-09-11:
OS: Windows 10
Version:1.02.027r

ccli0, 2024-09-11:
Modify the gcc_arm.ld file for flash and RAM size according to the specifications of your target chip.
Recompile and enter debug mode.
The ELF file must be converted to a BIN file or HEX file for flashing.
[image]

closed 2024-10-07 by prometheans152