OpenNuvoton/M480BSP #45: 483芯片的高速usbd如何强制为全速运行

Alg-tech, 2024-11-11:
目前我是强制设置HSUSBD->OPER=0x00
但是HID轮训间隔设置为1的时候，usb上报始终保持为2ms，不是期望的1ms的间隔
是我配置的不对么

ychuang3, 2024-11-12:
USB Device 是被動的，interval 由 USB Host 決定，決定的依據是 USB Device endpoint descriptor 的 bInterval  欄位值。
請在 USB Host 端 (應該是 Windows PC ?)， uninstall 這個 device 再重新 install，或者更換 USB VID/PID。
Windows 有可能記住第一次連接時的 interval 設定，沒有採用後來修改的設定。

Alg-tech, 2024-11-12:
你们有相关示例代码么

ychuang3, 2024-11-12:
M480 BSP 的 USB_HID_xxx sample code 都是 for USB 1.1 (full-speed) HID example
請自行修改 endpoint descriptor

Alg-tech, 2024-11-12:
我还想确认一下  当我设置HSUSBD->OPER=0x00 ，是否意味着不论当前的主机是否是全速还是高速 ，芯片都是全速usb

ychuang3, 2024-11-12:
M483KGxxx 系列沒有 HSUSBD，只有 USBD。
BSP 是 M487 系列與 M483 共用的，HSUSBD_xxx sample 只能運行在 M487
所以 program HSUSBD->OPER 時無效的
請先確認您使用的 IC 型號，您應該跟代理商提出需求。

Alg-tech, 2024-11-12:
我用的是M483SIDAE芯片，代理商给我说的意思是，芯片运行全速还是高速取决于主机端，没办法强制只在全速下运行

ychuang3, 2024-11-12:
M483SIDAE 是具備 USB 2.0 與 USB 1.1 兩個 USB port 的
如果您的目的是要執行在 full-speed, 那麼使用 USB 1.1 port 就可以了。 
HSUSBD->OPER 控制 high/full speed 沒錯，但是要看甚麼時間點去設定。

我不大了解您的需求，為何要跑在 full-speed 的原因，如果可以說明原由，或許有其他更好的作法。

Alg-tech, 2024-11-12:
好的 ，我先试试

Alg-tech, 2024-11-15:
M483SIDAE 是具備 USB 2.0 與 USB 1.1 兩個 USB port 的
如果您的目的是要執行在 full-speed, 那麼使用 USB 1.1 port 就可以了   

这计划的意思
是我既可以使用HUSBD 也可以使用FUSBD的库么，目前使用FUSBD并不能正常运行

ychuang3, 2024-11-15:
HSUSBD (USB 2.0)，USBD (USB 1.1)，這是兩個各自獨立的 IP。各自有一個 USB device port。
BSP 範例 HSUSBD_xxx 與 USBD_xxx 是分屬於不同 IP，彼此無關
HSUSBD_xxx 範例中沒有 full-speed 範例，全都是 high-speed
 
全部的 BSP 範例 USBD_xxx ， 都可以在新唐的開發板上面運行 full-speed USB。
如果還有問題，請詢問代理商。