OpenNuvoton/M480BSP #47: Hubs compatibility

jeromeDms, 2024-12-17:
Hi 
I'm using the M482 in a USB MIDI host product.
Everything works as expected but I'm facing one issue with some hubs.
Some are working, some are not working.

- If I connect the hub to the M482 host **with no connected device to the hub, it is recognized and connecting** devices one by one works fine, they are all recognized.
- If I connect the hub to the M482 host **with already connected devices to the hub, the hub is recognized but the connected devices do not trigger an interrupt**, and the host does not see the connected devices.

hub_status_irq() is called only one time but sc_bitmap is 0 after this first call (sounds like utr->buff[I] are only zero's), that results in no status change, thus no further detection of connected devices, since the itr is not rearmed in hub_polling

What about checking if utr->buff[I] contains only zeros in hub_status_irq() and re-submit interrupt-in transfer in that case ?

Any tip would be appreciated 
Thanks 
Jerome

jeromeDms, 2024-12-17:
Replying to myself : I modified the hub driver as follow and now it works :

```
static void hub_status_irq(UTR_T *utr)
{
    HUB_DEV_T   *hub;
    int         i;

    // HUB_DBGMSG("hub_status_irq - %d\n", utr->xfer_len);

    hub = (HUB_DEV_T *)utr->context;

    if (utr->status != 0)
    {
        USB_error("hub_status_irq - has error: 0x%x\n", utr->status); 
        return;
    }

		**wasDummyIrq = 1; // JDS V3.0**
		
    if (utr->xfer_len)
    {
        for (i = 0; i < utr->xfer_len; i++)
        {
            hub->sc_bitmap |= (utr->buff[i] << (i * 8));
						**if(utr->buff[i] != 0) wasDummyIrq = 0; // JDS V3.0**
        }
       
				// HUB_DBGMSG("hub_status_irq - status bitmap: 0x%x\n", hub->sc_bitmap); 
    }
}
```

then in hub_polling()

```
static int  hub_polling(void)
{
    HUB_DEV_T   *hub;
    UTR_T       *utr;
    int         i, ret, port, change = 0;

    if (_hub_polling_mutex)                 /* do nothing                                 */
        return 0;

    _hub_polling_mutex = 1;

    for (i = 0; i < MAX_HUB_DEVICE; i++)
    {
        if ((g_hub_dev[i].iface != NULL) && ((g_hub_dev[i].sc_bitmap) **|| (wasDummyIrq == 1)**)) // JDS V3.0
        {
            /*
             *  This hub device has status change
             */
            hub = &g_hub_dev[i];
            
						**if(g_hub_dev[i].sc_bitmap) // JDS V3.0 add this if as we could be here only from wasDummyIrq**
							change = 1;
						**else
							wasDummyIrq = 0;**

            HUB_DBGMSG("HUB [%s] hub status change 0x%x.\n", hub->pos_id, hub->sc_bitmap);

            if (hub->sc_bitmap & 0x1)
                hub_status_change(hub);

            for (port = 1; port <= hub->bNbrPorts; port++)
            {
                if (hub->sc_bitmap & (1 << port))
                {
                    ret = port_status_change(hub, port);
                    if (ret < 0)
                        break;
                }
            }
            hub->sc_bitmap = 0;
            /* re-submit interrupt-in transfer */
            if (ret == 0)
            {
                utr = hub->utr;
                utr->xfer_len = 0;
                ret = usbh_int_xfer(utr);
                if (ret)
                {
                    USB_error("Failed to re-submit HUB [%s] interrupt-in request (%d)", hub->pos_id, ret);
                }
            }
        }
    }
    _hub_polling_mutex = 0;
    return change;
}
```

ZYCHEN0, 2025-01-16:
Test device : M487KMCAN
Test code : HSUSBH_USBH_HID sample code 
Test step :
	Test A:
		1.Connect two low-speed devices and two full-speed devices to the hub
		2.Connect the hub to the board
	Test B:
		1.Connect the hub to the board
		2.Connect two high-speed devices and two low-speed devices to the hub in sequence.
Test result: In both Test A and Test B, all devices were successfully enumerated.