OpenNuvoton/M480BSP #49: Using UART Rx and Tx

jeromeDms, 2025-01-08:
Hi
Is it possible to use UART Rx and Tx that are not grouped ?
I have some trouble to receive data on UART1 RX, when defined as follow.

The below example is using UART1 RX pin PC8 and UART1 TX pin PE13
For instance is this possible ?

// Set GPC multi-function pins for UART1 RXD PC8
SYS->GPC_MFPH &= ~(SYS_GPC_MFPH_PC8MFP_Msk);
SYS->GPC_MFPH |= (SYS_GPC_MFPH_PC8MFP_UART1_RXD);

// Set GPE multi-function pins for UART1 TXD PE13
SYS->GPE_MFPH &= ~(SYS_GPE_MFPH_PE13MFP_Msk);
SYS->GPE_MFPH |= (SYS_GPE_MFPH_PE13MFP_UART1_TXD);

Thanks

ychuang3, 2025-01-09:
This configuration seems fine. Are you not receiving any data at all on UART1 RX, or is there another issue?

jeromeDms, 2025-01-09:
I'm at the beginning of debugging the UART, I just wanted to know if It was possible to use the UART Rx Tx as described.
Thanks for the answer.