OpenNuvoton/M480BSP #51: Issue with USB Host and D+/D- pulldown resistors

jeromeDms, 2025-04-16:
Hi All
I've a USB Host project based on M482LGCAE, where we added pulldown resistors on the USB D+/D- data lines.
All work fine except with some devices, that are not working when directly connected to the M482 and working when connected through a USB hub.
Compared to the evaluation board NuMaker M483, the devices work when directly attached to the M483.
The main difference are the pulldown resistors that are not used in the NuMaker board.

What I found is when the device is detected in the ohci_rh_polling() function, the call to connect_device() results in the following difference when comparing our board (with pulldown resistors) to the NuMaker board :

In connec_device() from usb_core.c :

delay_us(100 * 1000);                   /* initially, give 100 ms delay               */
usbh_get_device_descriptor(udev, &udev->descriptor);
reset_device(udev);
delay_us(100 * 1000);
ret = usbh_set_address(udev);

When I set a breakpoint before calling reset_device(), the device seems unconnected in our board, while it is still connected in the NuMaker.

Our board :

[image]

NuMaker board :

[image]

I would be happy to read some tips of what can be done here, as I cannot remove the pulldown resistors on existing products.
I tried to re-reset the USB device by disabling alternate functions on D-/D+ lines, putting those lines low for 20ms and re-enabling the alternate function, without success.

Thanks for any tips you can provide.

ychuang3, 2025-04-17:
Judging from the port status, it is in a disconnected state. This is determined by hardware, and software cannot override it. Modifying the board circuitry seems unavoidable; please consider replacing it with a 0-ohm resistor.

jeromeDms, 2025-04-17:
Thanks.
We cannot modify the boards.
Also removing the pulldown seems the solution but again we have some boards already shipped to customers.
From your point of view, do you think the 15K pull-down resistors on D+/D- are the cause of the problem ?
I think the USB phy has its own pull-down resistors right ?
Any workaround to disable them ?

ychuang3, 2025-04-17:
The SYS_USBPHY register can control some settings of the USB 1.1 PHY, but it does not include any configuration related to the pull-down resistor, so it doesn't seem to resolve your issue.
Since the hardware determines the state as disconnected, I can't think of any workaround to force it into a connected state.